esphome/esphome #8: Main stream hardware

HydrelioxGitHub, 2018-04-18:
Hi, 
I'm very interested in your project. I'm wondering if esphomeyaml can be used with common hardware like sonoffS20 plug or arilux al-lc03 led driver ?

I'm an intensive user of home Assistant (and sometimes a contributor) and I'm looking for an easy way to convert and update my hardware. For now I'm using the tasmota firmware.

OttoWinter, 2018-04-21:
Sorry for the late reply, there were just lots of things going on... Anyway, the sonoffS20 should already be supported as its just using the ESP8266 chip and some GPIO pins for the power control. (*)

The arilux light would probably theoretically also be supported with the ESP32 using the [IR Transmitter Component](https://esphomelib.com/esphomeyaml/components/switch/ir_transmitter.html). Though the 2 problems would be 1. only ESP32 support for now (IR support for the ESP8266 is on my list) and 2. having the IR command codes built-in. I think I'll try to work on this integration together with the FastLED light support, as they both would require at least a little bit of changes in the light core.

(*) I seriously do not recommend using the Sonoff devices. From an electronics standpoint, they seem 
very dangerous to me. Much to little and/or no isolation between mains and logic levels, mains wires just soldered on and so on.

HydrelioxGitHub, 2018-05-02:
Thank you,

I have some Sonoff switch (my S20 plug are already used) to make some experiment with your firmware. 

Thank you also for your advice for sonoff devices. I will be more careful for my next purchases. Is there a general weak conception, or is it dangerous with only powerful devices (with more device that need more than 2 or 3 amps) ?

OttoWinter, 2018-05-04:
So my previous comment was probably worded a bit too harsh. The Sonoff S20 is a fine device and can be used. I'm just worried with unexperienced (note: in general) people opening up their devices with mains voltage and attaching some wires for uploading. I'm sure one day this will lead to someone getting seriously injured. The second reason is that these china-ware devices usually don't follow many safety standards (soldered on mains wires, insufficient isolation between logic).

Looking more closely at the Sonoff S20, it's probably fine to use. At least from images I pulled from the images it doesn't look too bad after all, but I still personally wouldn't use them in my house.

Regarding the power of the devices, the S20 can probably handle 2-3 amps with its internal relay, but I don't have any specific data.

Regards,
Otto

Landrash, 2018-05-17:
Think this one can be closed now :+1:

OttoWinter, 2018-05-27:
@simonszu You know much more about this than I do - So thank you very much for your comment on this. And you're right that it's probably much safer to just buy this sort of product than attempting to do so on your own 100%!

HydrelioxGitHub, 2018-06-01:
With 1.6 release and sonoff documentation, I think this issue should be closed.

closed 2018-06-01 by HydrelioxGitHub