esphome/esphome #9: DHT sensor GPIO pin is not properly set

simonszu, 2018-04-24:
I have some problems with setting up a DHT22 sensor. After making sure that the hardware is fine, it is definitely a software issue. I dug into some files, and noticed the following. This is the sensor definition in the YAML:
```
sensor:
  - platform: dht
    pin: D4
    model: DHT22
    temperature:
      name: "Schlafzimmer Temperatur"
    humidity:
      name: "Schlafzimmer Luftfeuchtigkeit"
    update_interval: 15s
```
The pin ist set to D4, as this is the official D1 mini DHT22 shield. Unfortunately, after compiling the firmware, i only get `NaN` values, so i looked in the generated code. The DHT22 sensor is set up like following:
```
  auto dht_sensor = App.make_dht_sensor("Schlafzimmer Temperatur", "Schlafzimmer Luftfeuchtigkeit", 0, 15000);
  dht_sensor.dht->set_dht_model(DHT::DHT22);
```

If i interpret esphomelib's API documentation correctly, the third argument of `make_dht_sensor` should be the sensor's pin - which is set to 0, contrary to the YAML's definition of D4. The documentation of esphomeyaml looks like that one has to specify `D4` and not only `4`. However, i have noticed that, if i specify `4` as the pin, it is correctly parsed and written in the source file - but, of course, the ESP fails to communicate with the DHT due to the wrong pin. 

So, either i have not fully understood the config file format (especially after taking a look at `pins.py`) or there is a bug...?

OttoWinter, 2018-04-24:
esphomeyaml automatically translates the `D4`, `D0`, `D<x>` pins to the chipset GPIO numbering from the platform name. For example, the `D4` pin on the Wemos `D4` mini is internally known as `GPIO2`.

[image]
(the dark grey values are the GPIO pin numbers, and the `D<x>` numbers are the aliases by Wemos D1 mini)

It uses [these lists](https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h) by the awesome arduino espressif8266 projects to do so. Unfortunately, it seems like I had a small typo while copying the pin aliases from there into esphomeyaml: The pin `D4` maps to the internal `GPIO2` pin, not the `GPIO0` pin.

Thanks for reporting this 👍Copying those lists was quite a mess and I apparently had a typo there. I've now gone through the other pin aliases and have not seen this issue with other pins.

I'll probably release a new release of esphomeyaml in 1-2 weeks (my final exams start in 1,5 weeks, so I need to study 📚🤓). For now I'd just use `pin: 2` if it's connected to `D4` and `pin: 4` if the sensor is connected to `D2`.

simonszu, 2018-04-24:
Thanks for your fast feedback. I'll do so tomorrow. Have success with your finals.

commit 7915e420 (2018-04-24): Fix Wemos D1 Mini Pin Numbering (fixes #9)
closed 2018-04-24 by OttoWinter