esphome/esphome #10: Run code without wifi

Suhaib10, 2018-05-15:
First off thank you for your amazing work in creating this.

Secondly I have a slight issue/feature recommendation. I am using an esp8266 to control the led drivers powering my reef lights over mqtt using HA, but the issue is the wifi signal there is pretty weak. So if the Esp loses wifi it turns off the lights within seconds. 

So is it possible to just keep the last mqtt value until it gets wifi instead of turning the leds off.

OttoWinter, 2018-05-15:
Ah yes that is not very good.

The thing that's happening is that the MQTT client (or the WiFi client) see that their connection dropped and try to reconnect by just trying to re-connect. However, if this reconnection doesn't work within a set time period of 60 seconds or so, the node will reset itself.

The reason for this is that resetting the board has, especially on the ESP8266, almost magic effects and can fix almost all problems. So disabling the restart feature would probably cause some problems.

Possible work-arounds that could solve the problem (just thinking out loud here):

 - Make the time period to restart the device after a lost connection user-configurable. Problem with this is that it probably won't work with the ESP8266, since it for some reason it doesn't want to re-connect to WiFi after a lost connection without a full restart.
 - On the ESP32, I already have a flash preferences API set up. All ESP32s already store all light/switch/fan states in the flash area of the chip. When a restart happens, all states are automatically restored, so the lights won't turn off again. Maybe I could try porting that preferences API to the ESP8266... Would be quite a task though but also probably the best option.
 - Somehow use MQTT to restore states using the `retain` flags. Problem is that this probably won't work too well with MQTT lights because Home Assistant never sends the entire state.

This problem won't be too easy to solve, but it's definitely something that I should support/handle better. Thanks for the issue report!

OttoWinter, 2018-06-14:
So automations all those things now run without active WiFi and MQTT connections (since 1.7.0). 🎉

See https://esphomelib.com/esphomeyaml/guides/changelog.html#new-features

closed 2018-06-14 by OttoWinter