esphome/esphome #11: JSON string in MQTT Birth and LWT Payloads

bwicks, 2018-05-15:
Hi,

I’m trying to use a JSON string in Birth and LWT payloads but getting an error regarding 
Mqtt: - Invalid config for [mqtt]: string value cannot he dictionary or list.

Is there some way of getting this to work?

bwicks, 2018-05-15:
Here's my mqtt yaml for reference:

mqtt:
  broker: '192.168.1.143'
  username: 'home'
  password: 'kit'

  birth_message:
    topic: homebridge/to/set
    payload: {
 "name": "smokealarm1",
"service_name": "SmokeSensor",
 "characteristic": "SmokeDetected",
"value": 1
}
    qos: 1

  will_message:
    topic: homebridge/to/set
    payload: {
"name": "smokealarm1",
"service_name": "SmokeSensor",
"characteristic": "SmokeDetected",
"value": 0
}
    qos: 1

OttoWinter, 2018-05-15:
Your JSON payload is being interpreted as a YAML mapping (like when you write `broker: '192.168.1.143'`). So esphomeyaml receives the configuration like this:

```yaml
mqtt:
  broker: '192.168.1.143'
  username: 'home'
  password: 'kit'
  birth_message:
    topic: homebridge/to/set
    payload:
      name: "smokealarm1"
      service_name: "SmokeSensor"
      characteristic: "SmokeDetected"
      value: 1
    qos: 1

  will_message:
    topic: homebridge/to/set
    payload:
      name: "smokealarm1"
      service_name: "SmokeSensor"
      characteristic: "SmokeDetected"
      value: 0
    qos: 1
```

The payload parameter, however, only accepts strings (like `'{"name":"smokealarm1", ...}'`).

So your configuration should look like this:

```yaml
mqtt:
  # ...
  birth_message:
    topic: homebridge/to/set
    payload: >-
      {
        "name": "smokealarm1",
        "service_name": "SmokeSensor",
        "characteristic": "SmokeDetected",
        "value": 1
      }
    qos: 1
  will_message:
    topic: homebridge/to/set
    payload: >-
      {
        "name": "smokealarm1",
        "service_name": "SmokeSensor",
        "characteristic": "SmokeDetected",
        "value": 0
      }
    qos: 1
```

The `>-` keeps the YAML parser from interpreting the value as a dictionary and makes it use everything indented in there as a string.

However, I just noticed above payload does not work because I don't escape the payload parameter when generating C++ code. I've now fixed the bug and it will be in the next esphomeyaml release. For now, you can do this

```yaml
mqtt:
  # ...
  birth_message:
    topic: homebridge/to/set
    payload: "{\\042name\\042:\\042smokealarm1\\042,\\042service_name\\042:\\042SmokeSensor\\042,\\042characteristic\\042:\\042SmokeDetected\\042,\\042value\\042:1}"
    qos: 1
  will_message:
    topic: homebridge/to/set
    payload: "{\\042name\\042:\\042smokealarm1\\042,\\042service_name\\042:\\042SmokeSensor\\042,\\042characteristic\\042:\\042SmokeDetected\\042,\\042value\\042:0}"
    qos: 1
```

closed 2018-05-15 by OttoWinter

bwicks, 2018-05-15:
Wow, such a prompt response.
Do you have a patreon or donation page?

OttoWinter, 2018-05-16:
Well, I was just working on esphomeyaml while you were writing so I just quickly jumped over to Github and wrote a response.

About the donations: I'm already very happy when people value my work. Currently, I don't really want to setup donations or anything like that because a) then it would feel like "work" and not a fun side project and b) I've never really donated to other people that much myself in the online community, so I can't really justify accepting them for myself :P

commit 9979ee6d (2018-05-15): Fix C++ String escaping fixes #11