esphome/esphome #13: Current Docker image generates dependency on esphomelib.git#v1.4.0

apeeters, 2018-05-16:
Hi,

Apparently the current Docker image generates a PlatformIO dependency on
https://github.com/OttoWinter/esphomelib.git#v1.4.0 instead of 1.5.0.

$ docker pull ottowinter/esphomeyaml
Using default tag: latest
latest: Pulling from ottowinter/esphomeyaml
Digest: sha256:a0698ee356ea1335ea50a68ee9c9dbdbdddace906c47e85ab12fd167e3added3
Status: Image is up to date for ottowinter/esphomeyaml:latest

OttoWinter, 2018-05-16:
Oh 😅 thanks for the info! Apparently I did the [github release](https://github.com/OttoWinter/esphomeyaml/releases/tag/v1.5.0) before pushing the changes in 1.5.0

I'm going to release a small bugfix release 1.5.1 today. And hopefully I'll do the release right this time.

OttoWinter, 2018-05-17:
Fixed with 1.5.1 (and 1.5.2) 👍 Thanks again for reporting all these bugs!

closed 2018-05-17 by OttoWinter