esphome/esphome #19: error during compilation when Installing toolchain-xtensa32

oncleben31, 2018-05-24:
Trying to set my first sensor with esphomeyaml, I'm creating a dallas component base yaml file.

When trying to compile, I have the following error:
```
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Error: Could not find a version that satisfies the requirement '~2.50200.0' for your system 'linux_armv7l'
```

yaml file: 
```
esphomeyaml:
  name: capteur_piscine
  platform: ESP32
  board: lolin32

wifi:
  ssid: '####'
  password: '#####'

mqtt:
  broker: '192.168.1.28'
  username: '####'
  password: '####'

# Enable logging
logger:

ota:
  password: 'Konishiwa24'

# Example configuration entry
dallas:
  - id: dallas_hub1
    pin: 4

# Individual sensors
sensor:
  - platform: dallas
    dallas_id: "dallas_hub1"
    address: 0x28FF88A9C117049C
    resolution: 10
    name: "Température Piscine"
```

Hardware/Software: 

- Raspberry Pi 3
- HassIO 0.69.1

OttoWinter, 2018-05-25:
Yes that's because the build toolchain for the ESP32 is currently not available for RPi's armhf and similar architecture. (The setup guide mentions behind the main Button in the dashboard mentions this, albeit in a quote hidden way.)

I tried to compile the compiler for the RPi myself (and not download a pre-built one), but some of the compiler's code simply doesn't work with the RPi. So currently I don't really have a fix for this and you need to Instanz esphomeyaml on your computer to use the ESP32.

oncleben31, 2018-05-25:
Ah too bad. 
I tried to look a second time and I didn't see the mention about this. Perhaps you should try to make it more visible.
I will try from my Mac.

oncleben31, 2018-05-25:
It seems some succeed to compile the toolchain. https://github.com/platformio/platform-espressif32/issues/35
Not sure I'm not an expert. Perhaps you can check ?

oncleben31, 2018-05-25:
Other good news it seems someone has uploaded the arm version of the toolchain. I've just tried to relaunch the compilation and I unblock more steps but still with errors at the end.
```
INFO [esphomeyaml.__main__] Generating C++ source...
INFO [esphomeyaml.__main__] Compiling app...
INFO [esphomeyaml.__main__] Running:  platformio run -d /config/esphomeyaml/capteur_piscine
[Fri May 25 22:53:33 2018] Processing capteur_piscine (platform: espressif32; board: lolin32; framework: arduino)
--------------------------------------------------------------------------------
Warning! Library `{'requirements': None, 'name': 'ESPmDNS'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'ESPmDNS'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Warning! Library `{'requirements': None, 'name': 'Preferences'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Preferences'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Warning! Library `{'requirements': None, 'name': 'Update'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Update'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Downloading
Downloading
Unpacking
Unpacking
PackageManager: Installing framework-arduinoespressif32 @ ~1.4.0
Downloading
Downloading
Unpacking
Unpacking
PackageManager: Installing tool-esptoolpy @ ~1.20100.0
Downloading
Downloading
Unpacking
Unpacking
CorePackageManager: Installing tool-scons @ ~2.20501.4
Downloading
Downloading
Unpacking
Unpacking
tool-scons @ 2.20501.4 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > WEMOS LOLIN32
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 24 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoOTA> v1.0
|   |-- <WiFi> v1.0
|   |-- <ESPmDNS> v1.0
|   |   |-- <WiFi> v1.0
|   |-- <Update> v1.0
|-- <ESPmDNS> v1.0
|   |-- <WiFi> v1.0
|-- <Preferences> v1.0
|-- <Update> v1.0
|-- <esphomelib> v1.5.3 #2693649
|   |-- <AsyncMqttClient> v0.8.2
|   |   |-- <AsyncTCP> v1.0.1
|   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <ESP Async WebServer> v1.1.1
|   |   |-- <AsyncTCP> v1.0.1
|   |   |-- <WiFi> v1.0
|   |   |-- <FS> v1.0
|   |   |-- <ArduinoJson-esphomelib> v5.13.3
|   |-- <FastLED> v3.1.8 #d416194
|   |-- <WiFi> v1.0
|   |-- <ESPmDNS> v1.0
|   |   |-- <WiFi> v1.0
|   |-- <ArduinoOTA> v1.0
|   |   |-- <WiFi> v1.0
|   |   |-- <ESPmDNS> v1.0
|   |   |   |-- <WiFi> v1.0
|   |   |-- <Update> v1.0
|   |-- <Preferences> v1.0
[... 22 lines cut ...]
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdaPvj: symbol not found
*** [.pioenvs/capteur_piscine/lib603/FastLED/bitswap.cpp.o] Error 127
Error loading shared library libstdc++.so.6: No such file or directory (needed by /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++)
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_memory_used: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znwj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdlPvj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znaj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdaPvj: symbol not found
*** [.pioenvs/capteur_piscine/lib603/FastLED/colorpalettes.cpp.o] Error 127
Error loading shared library libstdc++.so.6: No such file or directory (needed by /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++)
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_memory_used: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znwj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdlPvj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znaj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdaPvj: symbol not found
*** [.pioenvs/capteur_piscine/lib603/FastLED/colorutils.cpp.o] Error 127
========================= [ERROR] Took 124.77 seconds =========================
 ```

OttoWinter, 2018-05-27:
Oh that's just awesome! Thank you very much! (What a coincidence, I was looking at that exact issue page about a week ago, and the comment didn't exist back then 😀)

About the `libstdc++.so.6` issue: I will check it out when trying to incorporate ESP32 for the HassIO add-on, but [this issue](https://github.com/platformio/platformio-core/issues/503) seems related

Flipper84, 2018-09-08:
how do i install esphomeyaml on my desktop? running windows, or on macbook??

oncleben31, 2018-09-17:
> how do i install esphomeyaml on my desktop? running windows, or on macbook??

I succed on my macbook following the instructions: https://esphomelib.com/esphomeyaml/guides/getting_started_command_line.html

OttoWinter, 2018-11-27:
Ok, so good news: compiling for ESP32s is now possible on RPis. The aarch64 issue is tracked here: https://github.com/OttoWinter/esphomeyaml/issues/81

closed 2018-11-27 by OttoWinter