esphome/esphome #20: Feature request: Specify custom target IPs for OTA update

simonszu, 2018-05-26:
Following some best practices, i have separated my IoT Devices into a separate VLAN, which also has a separate SSID in my Wifi access point. I don't want to impute writing malicious code to you, @OttoWinter , but it was easier for me to put the ESPs into this VLAN as well, for firewall ruling and similar actions. 

To cut a long story short: Another VLAN is another broadcast domain, so the resolving of <hostname>.local during OTA process is failing. The ESP host itself is pingable via its IP address and even via internal DNS, since i specified an entry for it. So it would be great to have an option in the YAML file to specify a specific IP address or hostname for OTA, so that esphomeyaml isn't sending the invitation to <hostname>.local but merely to the specified IP address. 

If one does not explicitly specify this option, trying <hostname>.local would be a good fallback option. 

On a somehow related side note: Is port 13564 for the OTA server to listen on a fixed value? I need to specify a firewall rule to enable talkback from the IoT VLAN to the VLAN where my desktop PC resides.

OttoWinter, 2018-05-27:
Jep, I understand. Support for customising `.local` has already been achieved with @HeMan's PR #16 and will be part of 1.6.0

Another option that I myself am using is to specify a static IP for the node, the OTA process will automatically use that one instead. See https://esphomelib.com/esphomeyaml/components/wifi.html (It also improves the time until a connection is established by a lot. I don't know much about networking, but somehow the DHCP requests are quite slow)

For OTA: The ESP always uses a specific port for accepting OTA requests: 3232 for the ESP32 and 8266 for the ESP8266 (these are the default ArduinoOTA values - this can be customised using `port:` option [here](https://esphomelib.com/esphomeyaml/components/ota.html)).

The host will choose a random port for sending the upload file. This, however, can also be customised using the `--host-port` option for the run/upload commands:

```bash
esphomeyaml config/livingroom.yaml run --host-port 8000

# Using a special environment variable:
ESPHOMEYAML_OTA_HOST_PORT=8000 esphomeyaml config/livingroom.yaml run
```

carlosmgr, 2018-06-07:
OttoWinter this is nightmare for firewalls... lol 

Security it's my job, i will try to explain a bit of network/firewall  

So, the ESP it's listening on port 8266 and responds back in a high random port!
After the response the host sends the file on one other random port! 

Every (correctly configured) firewall will block this type of behavior.

By design a firewall only permits responses from a ip/port/protocol that "was" permitted to pass when the communication was initiated.
If esp is listening on one port it must respond on the same port, otherwise it's a new connection and the firewall will block it

if it's done in this way you can secure the host/server that sends the file, because the 8266 port its closed to incoming communications, and the esp only answers back if we have a password. 

So, if you use 8266 port the esp must respond to the same port
If you use one other port to send the file, the esp must respond for the same port. This port must be standardized the same way 8266 was.  
 
For example MQTT uses port tcp 1883 or 8883, the node/broker sends and responds to that port.

OttoWinter, 2018-06-07:
Ok, to clarify: for the host where the OTA update is sent to, there is this behavior (which I think already covers your use case):
 * If nothing is specified, OTA updates are sent to `<NODE_NAME>.local`
 * If a "domain" option in the WiFi section is specified uploads happen to `<NODE_NAME><DOMAIN>`
 * If a `manual_ip` is specified in the WiFi section, uploads are sent to that static IP address.
 * If you give esphomeyaml on the command line an `--upload-port` argument, you can manually override all of that.

Concerning the firewall horror 😉: I don't even know myself how the OTA mechanism works behind the scenes. I literally just copied over the OTA code from the Arduino project and made it "work" with esphomeyaml. I mean AFAIK they've been using this weird random port selection for years now...

But I think using the `ESPHOMEYAML_OTA_HOST_PORT=8000` you can pretty much customize everything about the OTA mechanism if your firewall blocks it.

carlosmgr, 2018-06-07:
I actually have define a "node_name.domain" and a "manual_ip".
How can i define the "ESPHOMEYAML_OTA_HOST_PORT=8000" on the dashboard?

OttoWinter, 2018-06-07:
You *should* be able to do that within the HassIO add-on options:

[image]

However, that doesn't seem to work for me and I'm not sure why (maybe it works for you?) However, as this is HassIO and all internal ports are pretty much statically defined inside the `config.json`, I don't think there's a way around it if changing the port mapping doesn't work...

aequitas, 2018-06-07:
I have a similar network setup (separate vlan for robots) and ran into the same issue. I separate them in two problems:

- due to lack of zeroconf broadcast an explicit ip needs to be specified during upload
- the port the host uses to receive the device response after OTA invitation is random generated

The second problem is solved by the `--host-port` argument. And should work when a static ip is configured in the yaml file. However if you do not want to use static ip's it is possible to pass `--upload-port` which can be a hostname or ip address of the device. However current logic in esphomeyaml causes the `--host-port` argument to be dropped because `platformio run` is used for upload instead of `espota.py`. The former does not ever support the port argument afaict.

I made a change which should support using both arguments (and probably the environment variable mentioned above, haven't tested that). 

https://github.com/OttoWinter/esphomeyaml/pull/36

After this you can open a specific port on the firewall for ota responses and upload to a specific ip using: `esphomeyaml zonnescherm.yaml upload --upload-port 10.0.0.55 --host-port 8000`

carlosmgr, 2018-06-07:
Thanks the option "ESPHOMEYAML_OTA_HOST_PORT=8000" worked

OttoWinter, 2018-11-27:
Should no longer be an issue in 1.9 - esphomeyaml will instruct the user how to do this if name resolution fails.

closed 2018-11-27 by OttoWinter

trp79, 2019-06-03:
Hi,

I have a similar setup with a IoT VLAN. I use Hass.IO with the ESPHOME add-on. I have a static IP configured in the esphome yaml for my esp32 board (esp32_doorbell.yaml). I can however not seen any instruction in esp32_doorbell.yaml in relation to name resolution or setting. As well I cannot see where to set the host port to e.g. 8000. Only the dashboard server port seems to be configurable (default 6052) - https://github.com/esphome/hassio/tree/master/esphome.

So, I get below error when trying to use OTA.

`INFO Successfully compiled program.
INFO Connecting to 10.0.30.50
ERROR Connecting to 10.0.30.50:3232 failed: [Errno 113] No route to host`