esphome/esphome #21: Program fails to run

tonycwhite, 2018-05-30:
IM running Home Assistant on Raspberry PI, and Ive carried out the install, using the command 
`pip install esphomeyaml`

When i run '`esphomeyaml livingroom.yaml wizard` I recieve notification theat esphomeyaml cannot be found.

I know this is probably a basic issue, but its not one i know how to resolve.

OttoWinter, 2018-05-30:
Hi! Probably the installation failed or your python 2 installation is not linked ([python environments are complicated....](https://m.xkcd.com/1987/))

* What did `pip install esphomeyaml` output? Have you tried running `pip2 install esphomeyaml`? esphomeyaml currently requires Python **2** due to a dependency, so you need to use ``pip2`` to use Python 2.
* What does `python2 -m esphomeyaml -h` output?

tonycwhite, 2018-05-30:
output from python2 -m edsphomeyaml -h

`python2 -m esphomeyaml -h
usage: esphomeyaml [-h]
                   configuration
                   {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version}
                   ...

positional arguments:
  configuration         Your YAML configuration file.
  {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version}
                        Commands
    config              Validate the configuration and spit it out.
    compile             Read the configuration and compile a program.
    upload              Validate the configuration and upload the latest
                        binary.
    logs                Validate the configuration and show all MQTT logs.
    run                 Validate the configuration, create a binary, upload
                        it, and start MQTT logs.
    clean-mqtt          Helper to clear an MQTT topic from retain messages.
    wizard              A helpful setup wizard that will guide you through
                        setting up esphomeyaml.
    mqtt-fingerprint    Get the SSL fingerprint from a MQTT broker.
    version             Print the esphomeyaml version and exit.

optional arguments:
  -h, --help            show this help message and exit
`
from the pip command

`Successfully installed bottle-0.12.13 certifi-2018.4.16 chardet-3.0.4 click-5.1 colorama-0.3.9 colorlog-3.1.4 esphomeyaml-1.5.3 idna-2.6 lockfile-0.12.2 paho-mqtt-1.3.1 platformio-3.5.2 pyserial-3.4 pyyaml-3.12 requests-2.18.4 semantic-version-2.6.0 urllib3-1.22 voluptuous-0.11.1`

to confirm, should the full run command be as initially stated, or with python2 infront?

`python2 esphomeyaml livingroom.yaml wizard`

thanks for the prompt response!

OttoWinter, 2018-05-30:
The full command should usually just be `esphomeyaml livingroom.yaml wizard`.

However, with python's complicated environment setup, this command style does not always work. I think what's going on here is that in your case the python 2 executables path is not setup in your `PATH` environment variable.

I'm seeing the same thing happen over at Home Assistant very often - everybody has different python installations and even the packages installed through `apt` and similar sometimes seem to not do things the right way. So if something is not working as expected, I usually just tell people to use 

```bash
python2 -m <PROGRAM_NAME> ...
```
it works in a lot more cases than the shorthand

```bash
esphomeyaml ...
```

So I think in your case, you should be able to just run

```bash
python2 -m esphomeyaml livingroom.yaml wizard
```

tonycwhite, 2018-05-30:
boom - resolved

Many thanks!

OttoWinter, 2018-05-31:
Perfect!

closed 2018-05-31 by OttoWinter