esphome/esphome #22: BME280

fabaff, 2018-05-31:
Using a BME280 with the following configuration is not able to retrieve the humidity.

```yaml
i2c:
  sda: 21
  scl: 22
  scan: False

sensor:
  - platform: bme280
    temperature:
      name: "Temperature"
      oversampling: 16x
    pressure:
      name: "Pressure"
    humidity:
      name: "Humidity"
    address: 0x76
    update_interval: 15s
```
Debug:

```bash
[C][sensor.bme280:095]: Setting up BME280...
[C][sensor.bme280:096]:     Address: 0x76
[C][sensor.bme280:126]:     Temperature Oversampling: 16x
[C][sensor.bme280:127]:     Pressure Oversampling: 16x
[C][sensor.bme280:128]:     Humidity Oversampling: 16x
[C][sensor.bme280:129]:     IIR Filter: OFF
[...]
[D][sensor.bme280:177]: Got temperature=26.0°C pressure=952.6hPa humidity=0.0
[D][sensor.mqtt:038]: 'Temperature': Pushing out value 25.980000 with accuracy 1
[D][sensor.mqtt:038]: 'Pressure': Pushing out value 952.551331 with accuracy 1
[D][sensor.mqtt:038]: 'Humidity': Pushing out value 0.000000 with accuracy 1
```

Sorry, I have only one of those sensors at hand at the moment.

OttoWinter, 2018-05-31:
Hmmm, that's no good :(

I will receive my own BME280 in 2-3 weeks and I will try to debug the issue then. In the meantime and if you have time, could you enable [`very_verbose` logs](https://esphomelib.com/esphomeyaml/components/logger.html#log-levels)? They show all the i2c data communicated and are amazing at diagnosing issues. Thanks!

fabaff, 2018-05-31:
With a different sensor it works fine. 

As a reference, the extended log:

<details>

```bash
[C][sensor.bme280:095]: Setting up BME280...
[C][sensor.bme280:096]:     Address: 0x76
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b11010000 (0xD0)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 1 bytes from 0x76:
[VV][i2c:133]:     Received 0b01100000 (0x60)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10001000 (0x88)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b01010101 (0x55)
[VV][i2c:133]:     Received 0b01101110 (0x6E)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10001010 (0x8A)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b11111101 (0xFD)
[VV][i2c:133]:     Received 0b01100110 (0x66)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10001100 (0x8C)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b00110010 (0x32)
[VV][i2c:133]:     Received 0b00000000 (0x00)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10001110 (0x8E)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b00000111 (0x07)
[VV][i2c:133]:     Received 0b10010001 (0x91)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10010000 (0x90)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b00000110 (0x06)
[VV][i2c:133]:     Received 0b11010110 (0xD6)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10010010 (0x92)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b11010000 (0xD0)
[VV][i2c:133]:     Received 0b00001011 (0x0B)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10010100 (0x94)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b01100100 (0x64)
[VV][i2c:133]:     Received 0b00100011 (0x23)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10010110 (0x96)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b11010000 (0xD0)
[VV][i2c:133]:     Received 0b11111111 (0xFF)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10011000 (0x98)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b11111001 (0xF9)
[VV][i2c:133]:     Received 0b11111111 (0xFF)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b10011010 (0x9A)
[... 108 lines cut ...]
[VV][i2c:133]:     Received 0b11101110 (0xEE)
[VV][i2c:133]:     Received 0b10110000 (0xB0)
[VV][i2c:076]: Beginning Transmission to 0x76:
[VV][i2c:109]:     Writing 0b11111101 (0xFD)
[VV][i2c:081]:     Transmission ended. Status code: 0x00
[VV][i2c:103]: Requesting 2 bytes from 0x76:
[VV][i2c:133]:     Received 0b01110011 (0x73)
[VV][i2c:133]:     Received 0b01000001 (0x41)
[D][sensor.bme280:177]: Got temperature=27.4°C pressure=952.0hPa humidity=50.7
[V][sensor.sensor:022]: 'Temperature': Received new value 27.430000
[D][sensor.mqtt:038]: 'Temperature': Pushing out value 27.430000 with accuracy 1
[V][mqtt.client:193]: Publish(topic='simple_ambient_sensor/sensor/temperature/state' payload='27.4' retain=1)
[V][sensor.sensor:022]: 'Pressure': Received new value 951.996155
[D][sensor.mqtt:038]: 'Pressure': Pushing out value 951.996155 with accuracy 1
[V][mqtt.client:193]: Publish(topic='simple_ambient_sensor/sensor/pressure/state' payload='952.0' retain=1)
[V][sensor.sensor:022]: 'Humidity': Received new value 50.656250
[D][sensor.mqtt:038]: 'Humidity': Pushing out value 50.656250 with accuracy 1
[V][mqtt.client:193]: Publish(topic='simple_ambient_sensor/sensor/humidity/state' payload='50.7' retain=1)
[VV][component:107]: Running defer '':1 with interval=0 last_execution=0 (now=5618)
[VV][component:107]: Running defer '':2 with interval=0 last_execution=0 (now=5621)
[VV][component:107]: Running defer '':3 with interval=0 last_execution=0 (now=5626)
[VV][component:107]: Running interval '':0 with interval=10000 last_execution=1121 (now=11122)
[VV][component:107]: Running interval 'update':0 with interval=15000 last_execution=4294966984 (now=14692)
```
</details>

closed 2018-05-31 by fabaff