esphome/esphome #23: Support for BMP280

fabaff, 2018-05-31:
The [BMP280](https://www.bosch-sensortec.com/bst/products/all_products/bmp280) is the "cheaper/simpler" version of the BME280 which is already supported.

Thanks in advance.

OttoWinter, 2018-05-31:
I will have a look at it. Doesn't seem too different from the BMP180 and BME280, so integrating it should not be that hard for me :)

fabaff, 2018-05-31:
I bought a bunch of these sensors a while back before I found the HTU21D. For temperature-only things are BMP280 still useful, Ion't care much about the pressure :wink:.

Well, if I'm the only user then it doesn't make sense to integrate it.

fabaff, 2018-06-12:
I tried to use the BME280 and make a BMP280 out of it. Only partial success, readings are off because it seems that I'm not able to initialize the sensor properly.

UhtredTheBold, 2018-07-19:
I have a GY-21P Breakout which I understand is a BMP280 combined with SI7021 for temperature and humdity (although honestly I only care about temperature). 
It would be great to be able to use this with esphomeyaml.

RicardoMTac, 2018-08-01:
hi guys. Im trying to work with 02 bmp280 sensors, but only one has working. im using spi protocol. someone can help me?? im studying arduino, very unprofessional. I already tested the two sensors individually. are working perfectly. 
/***************************************************************************
  This is a library for the BMP280 humidity, temperature & pressure sensor
  Designed specifically to work with the Adafruit BMEP280 Breakout 
  ----> http://www.adafruit.com/products/2651
  These sensors use I2C or SPI to communicate, 2 or 4 pins are required 
  to interface.
  Adafruit invests time and resources providing this open source code,
  please support Adafruit andopen-source hardware by purchasing products
  from Adafruit!
  Written by Limor Fried & Kevin Townsend for Adafruit Industries.  
  BSD license, all text above must be included in any redistribution
 ***************************************************************************/

#include <Wire.h>
#include <SPI.h>
#include  <RTClib.h> //Incremento Taccola 10/04
#include <Adafruit_Sensor.h>
#include <Adafruit_BMP280.h>

#define BMP_SCK 13
#define BMP_MISO 12
#define BMP_MOSI 11 
#define BMP_CS 10
#define BMP2_SCK 08 //incremento taccola 31/07
#define BMP2_MISO 07 //incremento taccola 31/07
#define BMP2_MOSI 06 //incremento taccola 31/07
#define BMP2_CS 05 //incremento taccola 31/07

RTC_DS1307 RTC; // Incremento Taccol 10/04
//Adafruit_BMP280 bme; // I2C
Adafruit_BMP280 bme(BMP_CS); // hardware SPI
Adafruit_BMP280 bme2(BMP2_CS); //incremento taccola 31/07
//Adafruit_BMP280 bme(BMP_CS, BMP_MOSI, BMP_MISO,  BMP_SCK);

void setup() {
  Serial.begin(9600);
   Wire.begin();  // Incremento Taccola 10/04
   RTC.begin(); // Incremento Taccola 10/04
  RTC.adjust(DateTime(__DATE__, __TIME__)); // Incremento Taccola 10/04
  Serial.println(F("BMP280 test"));
  
  if (!bme.begin()) {  
    Serial.println("Could not find a valid BMP280 sensor, check wiring!");
    while (1);
  }
 if (!bme2.begin()) {
    Serial.println("Could not find a valid BMP280_2 sensor, check wiring!");
    while (1); //incremento taccola 31/07 
    }  
}
  

void loop() {
    DateTime now = RTC.now(); // Incremento Taccola 10/04
    Serial.print("Temperature = ");
    Serial.print(bme.readTemperature());
    Serial.println(" *C");
    Serial.print("Pressure = ");
    Serial.print(bme.readPressure());
    Serial.println(" Pa");
    Serial.print("Pressure = ");
    Serial.print(bme.readPressure()/100000);
    Serial.println(" Atm");
    Serial.print("Approx altitude = ");
    Serial.print(bme.readAltitude(1018.0)); // this should be adjusted to your local forcase
    Serial.println(" m");
    Serial.println("");
    Serial.print("Temperature = ");    //incremento taccola 31/07
    Serial.print(bme2.readTemperature());
    Serial.println(" *C");
    Serial.print("Pressure = "); //incremento taccola 31/07
    Serial.print(bme2.readPressure());
    Serial.println(" Pa");
    Serial.print("Pressure = "); //incremento taccola 31/07
    Serial.print(bme2.readPressure()/100000);
    Serial.println(" Atm");
    Serial.print("Approx altitude = ");
    Serial.print(bme2.readAltitude(1018.0)); // this should be adjusted to your local forcase
    Serial.println(" m");

   
 Serial.print(now.year(), DEC);               // Incremento Taccola 10/04
    Serial.print('/');
    if (now.month() < 10) {
      Serial.print('0');
    }
    Serial.print(now.month(), DEC);         // Incremento Taccola 10/04
    Serial.print('/');
    if (now.day() < 10) {
      Serial.print('0');
    }
    Serial.print(now.day(), DEC);             // Incremento Taccola 10/04
    Serial.print(' ');
    if (now.hour() < 10) {
      Serial.print('0');
    }
    Serial.print(now.hour(), DEC);             // Incremento Taccola 10/04
    Serial.print(':');
    if (now.minute() < 10) {
      Serial.print('0');
    }
    Serial.print(now.minute(), DEC);            // Incremento Taccola 10/04
    Serial.print(':');
    if (now.second() < 10) {
      Serial.print('0');
    }
    Serial.print(now.second(), DEC);          // Incremento Taccola 10/04
    Serial.println();
    Serial.println();
    delay(15000);
}

OttoWinter, 2018-11-27:
Support is done: https://esphomelib.com/esphomeyaml/components/sensor/bmp280.html

closed 2018-11-27 by OttoWinter