esphome/esphome #24: Missing tornado dependency for dashboard

brandond, 2018-06-01:
Just updated to 1.6.0 and tried to start the dashboard. Got a stack trace instead:

```
Traceback (most recent call last):
  File "/home/media/django/bin/esphomeyaml", line 11, in <module>
    sys.exit(main())
  File "/home/media/django/local/lib/python2.7/site-packages/esphomeyaml/__main__.py", line 437, in main
    return run_esphomeyaml(sys.argv)
  File "/home/media/django/local/lib/python2.7/site-packages/esphomeyaml/__main__.py", line 414, in run_esphomeyaml
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/home/media/django/local/lib/python2.7/site-packages/esphomeyaml/__main__.py", line 318, in command_dashboard
    from esphomeyaml.dashboard import dashboard
  File "/home/media/django/local/lib/python2.7/site-packages/esphomeyaml/dashboard/dashboard.py", line 30, in <module>
    class EsphomeyamlCommandWebSocket(tornado.websocket.WebSocketHandler):
NameError: name 'tornado' is not defined
```

Works fine after installing tornado.

OttoWinter, 2018-06-01:
That was quick :)

And yes, I forgot to put that in the changelog notes. The Getting started guide had that in there: https://esphomelib.com/esphomeyaml/guides/getting_started_command_line.html#bonus-esphomeyaml-dashboard

(The reason I didn't put tornado into the `setup.py` requirements is that it's quite a big dependency and not part of the "core" esphomeyaml - only a UI extension for now)

brandond, 2018-06-01:
Ahh, OK, Maybe catch an ImportError and print a nice warning rather than throwing a stack trace, for those of us that just read the release notes and don't go back to the docs?

OttoWinter, 2018-06-01:
I updated the release notes too, so that shoudn't hopefully be a problem. But I think you're right that it would be good to do a try except import 👍

commit 4fa7bc19 (2018-06-01): Warn about tornado. Fixes #24
closed 2018-06-01 by OttoWinter