esphome/esphome #25: Custom variables in lambdas

brandond, 2018-06-01:
Is there any way to declare and initialize variables that can be used by the lambda? It'd be easy enough to drop into the c++ code to do this, I'm wondering if there's a way to do it from pure YAML.

Use case
--------
I have a very simple garage door opener that works as follows:
* When closed, will open on a single button press
* When open, will close on two presses 1 second apart
* When moving, will stop on a single button press
* When stopped (only the following two motions are available):
  * If previously closing, will open on a single press
  * If previously opening, will close on two presses 1 second apart

I have a relay wired up to the button contacts, and have simple `close` and `open` scripts in Home Assistant that trigger relay toggles 1 second apart - once to open or twice to close.

I'd like to convert this over to the `cover` model. However, I need a small state machine to determine how to open or close the door given its previous direction of motion. This could probably be accomplished with a lambda and a variable to store which way it was last moving.

brandond, 2018-06-01:
It's possible this is too complex for YAML, and I should just write the state machine in c++ and then call it from the lambda - but I thought I'd ask.

OttoWinter, 2018-06-01:
Yeah I was also thinking about this and ultimately I decided that it would just be way to much effort to add this with YAML where a simple global C++ variable already should do the job fine.

In the end, the people with these more complex automations and templates will, in most cases, anyway be people that already know C++ a bit.

Maybe in the future, but it's not a big priority for me now

aequitas, 2018-06-08:
Maybe you can try using an `optimistic` template switch component as 'variable'. You can use this to store the state of movement, something like this:

```yaml
switch:
  - platform: gpio
    name: "press the button"
    id: the_button
    pin: 1
    discovery: false // internal variable

  - platform: template
    name: "Is moving"
    id: is_moving
    optimistic: true
    discovery: false // internal variable

cover:
  - platform: template
    name: "Door"
    optimistic: true
    open_action:
      - lambda: >-
          if (id(is_moving).value){
            // single press to stop moving
            id(the_button).write_state(true);
            delay(500); //pseudocode, do lambda's support delay?
            id(the_button).write_state(false);
            delay(500); //pseudocode, do lambda's support delay?
            // single press to open
            id(the_button).write_state(true);
            delay(500); //pseudocode, do lambda's support delay?
            id(the_button).write_state(false);
            delay(500); //pseudocode, do lambda's support delay?
          } else {
            // single press to open
            id(the_button).write_state(true);
            delay(500); //pseudocode, do lambda's support delay?
            id(the_button).write_state(false);
          }
          // store in 'variable' that the door is moving
          id(is_moving).write_state(true);
          // after the door has finished opening (1 minute), reset moving state
          delay(60*1000);
          id(is_moving).write_state(false);
```

OttoWinter, 2018-06-08:
I think it would be even cleaner with the binary sensor API: https://esphomelib.com/esphomeyaml/components/binary_sensor/index.html#lambda-calls

```yaml
binary_sensor:
  - platform: template
    lambda: 'return {};' # Just do nothing
    name: Some Name
    id: bool_value

in_some_lambda: >-
  id(bool_value). publish_state(true);

in_other_lambda: >-
  if (id(bool_value).value) {
    // do something
  } else {
    // do something else
  }
```

But once again, just putting a `bool my_value;` as a global variable in your `src/main.cpp` is IMO a much shorter and cleaner solution. esphomeyaml was intentionally made in such a way that users should be able to, without any issues, write code in the C++ source file...

brandond, 2018-06-08:
Yeah, it'd be nice to do it all from YAML though. Maybe if there were pre/post vars that we could put simple chunks of code into that were included as-is into the generated source?

OttoWinter, 2018-11-27:
See https://esphomelib.com/esphomeyaml/guides/automations.html#bonus-2-global-variables

closed 2018-11-27 by OttoWinter