esphome/esphome #26: Feature Request: Sensor for voltage of ESP

leeuwte, 2018-06-01:
Some of my ESP's are running on battery. It would be nice do add a sensor which returns the voltage of the power supply. 

More info on; https://arduino-esp8266.readthedocs.io/en/2.4.1/libraries.html?highlight=getVcc#esp-specific-apis

OttoWinter, 2018-06-01:
Jep will take a look at that. Only need to figure out what unit the value of that is in...

closed 2018-06-02 by OttoWinter

brandond, 2018-06-04:
There's a note in the docs:
> ESP.getVcc() may be used to measure supply voltage. ESP needs to reconfigure the ADC at startup in order for this feature to be available. Add the following line to the top of your sketch to use getVcc:
>    
> `ADC_MODE(ADC_VCC);`

Is that still WIP?

OttoWinter, 2018-06-04:
Unfortunately that was the only way of getting it working - the problem is that platformio's linking process is removing all unused symbols from statically linked libraries. And what `ADC_MODE` does is define a function (which is a weak symbol in the arduino core).

OttoWinter, 2018-06-04:
I don't really understand your question I think, but see https://esphomelib.com/esphomeyaml/components/sensor/adc.html#esp8266-measuring-vcc

brandond, 2018-06-04:
Ahh, that was the bit that I was missing! I saw https://github.com/OttoWinter/esphomelib/commit/a2b70582e07e62aa713fc50c467013b76e9fdc91 but was wondering where the ADC_MODE call was at. Sorry!

commit eb62599a (2018-06-02): ADC VCC support Fixes #26