esphome/esphome #30: Addon doesn't work when behind ssl nginx

ciotlosm, 2018-06-01:
I'm using bestlibre nginx to map ports to sub domains and have proper certificates. Unfortunately it seems the ws connection when compiling is not secure or not taking into account that it's running on secure when trying to connect back. 

I get a warning in chrome and then nothing happens when compiling. Is there a way to get around this?

ciotlosm, 2018-06-01:
Update: My Sonoff 4CH failed to start after update using the generated binary. It was previously flashed with Tasmota. 

I was able to restore it using FTDI.

OttoWinter, 2018-06-02:
Yeah I think at the moment, as there's no authentication yet, implementing https support would give a "false sense of security". Authentication is on my list of things to do, but I think most people will just flash their devices while they're in their local network anyway, so..

About the 4CH: That's not good :( - I just tried again with my own Sonoff 4CH, this time with first installing tasmota and then using the OTA update method and it worked fine for my setup 😕

If you're still willing to try this out, here are a few gotchas:

 * Make sure the WiFi and MQTT settings are right. Otherwise the device will just try connecting again and again.
 * Most FTDI adapters can't deliver enough current to feed the 4CH - I don't know your setup, but that was one problem I had initially.

ciotlosm, 2018-06-02:
Wifi and MQTT should have been fine. I made sure I double checked each parameter I entered in the wizard. I was running Tasmota 5.8 or 5.10 on the device. I could see the device get an ip address, but no web interface available and no device appeared in home assistant. Pressing buttons also didn't trigger relays. 

Would also be nice to have an interface to discover and manage devices like SonWeb for Tasmota

ciotlosm, 2018-06-02:
Thanks for the detailed explanation. Probably my hope to get something working out of the box using the wizard. Maybe my expectation was wrong. 

I'll try and give it another go on a sonoff as soon as I have a bit more time. I might ask for help on Discord.

closed 2018-06-02 by ciotlosm