esphome/esphome #32: iBeacon Support

timbuchwaldt, 2018-06-05:
It would be super awesome if `esphomeyaml` could announce itself as an iBeacon via BLE.
This would allow iOS (Owntracks, mostly) to track positions in greater detail

OttoWinter, 2018-06-05:
Uhm yes, seems relatively straighforward to add (note to self: just more or less copy [this](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/ble_adv/main/app_bt.c))

I'm just thinking a bit about how to make the esphomeyaml configuration format... On the one hand I would kind of like this feature to be part of the [`esp32_ble`](https://esphomelib.com/esphomeyaml/components/esp32_ble.html) hub. But that hub is already used for the scanning stuff - and combining both modes into a single configuration component would probably just confuse people...

I'm not a big bluetooth expert... Do you know if there are other popular advertising protocols other than iBeacon? If not, I think I would just create a simple `esp32_ble_ibeacon` component or so

timbuchwaldt, 2018-06-06:
Yeah, that's what I assumed, but my C skills are... a little rusty to say the least :D 

There is Google Eddystone (which is somehow similar, but of course a little diferent) - iBeacon is supported on Android as well it seems, as long as the app uses a thirdparty library for scanning.

OttoWinter, 2018-06-11:
Just tried to implement this over here https://github.com/OttoWinter/esphomelib/tree/ble-beacon because it would have been an awesome feature for an otherwise kind of dull 1.7.0 release :)

I had all the code ready and it compiled. But when I clicked upload, I got a nice message telling me that the ESP32 has too little space for my sketch 😕 Next I tried to remove everything to no avail. Even [espressif's own ibeacon demo](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/ble_adv/main/app_bt.c) produces a binary size of ~1.5MB - with almost all ESP32 boards [capping out at 1.25MB](http://docs.platformio.org/en/latest/platforms/espressif32.html#external-tools).

So I'm not sure how espressif meant for this to be used if their own example already runs out of sketch space 😅 Probably they have other development boards with more flash size, but as I've seen no such commercially available devices, I don't see how to solve this :(

timbuchwaldt, 2018-06-11:
Damn :D Ok, this would have been too cool :/ But I guess good that I haven't ordered a bunch of ESP32s yet. Thanks for your effort though

brandond, 2018-06-11:
Those PlatformIO docs are wrong. Most ESPs actually have 32mbit flash, with up to 128mbit supported by the chip although I'm not aware of any actually available on the market. Apparently the image size is a limitation of the OTA process:
https://esp32.com/viewtopic.php?t=1738

OttoWinter, 2018-06-11:
@brandond You sir, are 100% correct + a lifesaver👍Those docs are in fact wrong. I just tried uploading manually with esptool and the uploading worked! But it didn't work directly (just a `user code done` message), I had to create a [custom partition table](https://github.com/espressif/arduino-esp32/commit/856823ef195ef5c28516395356e8a30f8af8e917) and now it worked. Just fixed a minor bug and esphomelib has its first BLE iBeacon:

[image]

HydrelioxGitHub, 2018-06-11:
I'm not very aware of Esp32 features.... One device could be used to control some sensors AND be an ibeacon ?
Or the ibeacon feature must be on a dedicated device and used only for this purpose ?

OttoWinter, 2018-06-12:
@HydrelioxGitHub I mean you won't be able to use too many sensors with it due to flash size restrictions, but in principle: yes, you can. If you disable OTA (and the second OTA partition, a bit complicated), you could even have tons of sensors connected to it.

closed 2018-06-14 by OttoWinter