esphome/esphome #33: esphomeyaml Dashboard Authentication

carlosmgr, 2018-06-05:
Hi

Simple question!
Can you put some kind off authentication on esphomeyaml Dashboard ?

Thanks

carlosmgr, 2018-06-06:
It's understandable.

It's just that the dashboard it´s very helpful but everyone can see the authentication for the mqtt, wireless, OTA, etc, It's a big security weaknesses.

Maybe someone who wrote the HA authentication can help with this.

Congratulations and thanks for the esphomeyaml, this appears to have a big potential, i'm using espeasy and while it's easy it's at the same time limited and problematic (buggy). Also i have some 1MB wemos that i dont use because of the space occupied by espeasy vs OTA.

if esphomeyaml had the ability to control a motor with a motorshield and/or the possibility for control a program made in arduino for the esp would be very nice

carlosmgr, 2018-06-08:
You're the Man :)

I'll be out on business the next 2 days so i cannot test the authentication until then.

Yes, the idea was to use a custom Arduino code, i have to see the "Custom Sensor Component" and understand how it works.

About the motorshield: 
https://wiki.wemos.cc/products:d1_mini_shields:motor_shield
or 
https://hackaday.io/project/8856-incubator-controller/log/29291-node-mcu-motor-shield

Many thanks OttoWinter

OttoWinter, 2018-06-08:
I will have a look at that. Closing this issue as it's resolved

closed 2018-06-08 by OttoWinter