esphome/esphome #37: error with DHT

carlosmgr, 2018-06-07:
Hi 
I get the following error every time with the dht.

[21:37:51][C][sensor.mqtt:025]: Setting up MQTT Sensor '2'...
[21:37:51][C][sensor.mqtt:027]:     Expire After: 675s
Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 458, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 448, in run_esphomeyaml
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 320, in command_run
    return show_logs(config, args, port, escape=args.escape)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 216, in show_logs
    escape=escape)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/mqtt.py", line 65, in show_logs
    return initialize(config, [topic], on_message, username, password, client_id)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/mqtt.py", line 36, in initialize
    client.loop_forever()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1481, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1003, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1284, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1849, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2305, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2500, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2647, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/mqtt.py", line 63, in on_message
    print(time + message)
  File "/usr/local/lib/python2.7/dist-packages/colorama/ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "/usr/local/lib/python2.7/dist-packages/colorama/ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "/usr/local/lib/python2.7/dist-packages/colorama/ansitowin32.py", line 169, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "/usr/local/lib/python2.7/dist-packages/colorama/ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 68: ordinal not in range(128)

OttoWinter, 2018-06-07:
That's probably because the DHT sends the degree symbol (maps to `0xb0` ASCII character in your error log) over MQTT and the dashboard wasn't escaping it properly.

This issue *should* be fixed with v1.6.2 which I published yesterday. See https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-update-to-the-latest-version

carlosmgr, 2018-06-07:
Yes it is thanks

closed 2018-06-07 by carlosmgr