esphome/esphome #38: Feature request: with OTA update and node using deep sleep, wait until node wake up

oncleben31, 2018-06-07:
To simplify OTA updates when node is not easy reachable (for using usb or clicking reset button), let the commands `esphomeyaml config.yaml run/upload` wait at least the duration of deep sleep mode to be sure to catch the wake up window.

Alternative solution: using a mqtt message to disable deep sleep mode and wait for OTA.

OttoWinter, 2018-06-11:
Mhm, in a previous version of esphomelib (before I released it), I had a magic MQTT message that would 1. prevent the node from going into deep sleep and 2. make the upload wait until the node was active. Maybe I'll take a look at this at some point, currently I think it's more of a "issue with a workaround".

nickw444, 2018-06-13:
This would be a useful feature for myself! For now, hitting reset will do the trick.

oncleben31, 2018-06-13:
No need to rush but in the near futur my node will be place somewhere difficult to reach with an uptime of 15 secondes every hours. Difficult to catch ;-)

pyrosmiley, 2018-08-07:
As this gets looked into, is there a way to set deep sleep parameters over MQTT or even with a hardcoded schedule? I'm working on a solar powered project that I need to be fairly responsive at a few set points over the day, but overnight/midday, really doesn't need to be awake. Since battery life is my biggest fight right now, it'd be insanely cool if I could just set it with a large Deep Sleep time overnight, but a lower time during the day. Not even sure if that's possible?

nunofgs, 2018-08-07:
Just hit on this issue myself. I have a reed sensor with a 1s deep sleep and now I'm forced to flash it through USB since it doesn't have enough time to receive an OTA update.

@OttoWinter the functionality you described would be much appreciated.

pyrosmiley, 2018-08-19:
Yeah, even just that "magic MQTT message" would be amazing since HASS can send that before it needs to do stuff, then just send a `reset` when it's done. I'm actually getting really excited over the idea lol

OttoWinter, 2018-09-19:
The functionality I described will be possible with the next release using the `deep_sleep.prevent` action. The docs will also have an example for how to use this.

closed 2018-09-19 by OttoWinter