esphome/esphome #39: Deep Sleep settings: can't use run_duration and run_cycles in the same time

oncleben31, 2018-06-07:
According to documentation we can use bot parameter: 
```
To use deep_sleep first specify how long the node should be active, i.e. how long it should check sensor values and report them, using the run_duration and run_cycles options. If you use both in your configuration, any time either one of them is finished, the node will go into deep sleep mode.
```
In my case:
```
esphomeyaml:
  name: capteur_piscine
  platform: ESP32
  board: lolin32

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

mqtt:
  broker: '192.168.1.28'
  username: !secret mqtt_username
  password: !secret mqtt_password

# Enable logging
logger:

ota:
  password: !secret ota_password

# Example configuration entry
dallas:
  - id: dallas_hub1
    pin: 4
    update_interval: 1s

# Individual sensors
sensor:
  - platform: dallas
    dallas_id: "dallas_hub1"
    address: 0x9C0417C1A988FF28
    name: "Température Piscine"
    icon: "mdi:oil-temperature"
    expire_after: "1:02"
    availability:

deep_sleep:
  run_duration: 20s
  run_cycles: 15
  sleep_duration: 1h
```
Seems to not work, the node go to deep sleep before the first cycle.

OttoWinter, 2018-06-08:
So to clarify the thing in the docs: If you specify both `run_duration` and `run_cycles`, then the behavior is a follows: If either one of them is finished, then the node goes into deep sleep. So if the 15 loop cycles goes directly into deep sleep. Also if 20s are over, the node goes directly into deep sleep

In your case, probably `run_cycles: 15` is triggering the restart. Why does it not send the dallas values then?

The dallas sensors have a conversion time of about 750ms, during this time esphomelib goes through about ~50 or so loop cycles. At t=0, esphomelib makes the sensors start a measurement, and at t=750ms, esphomelib reads those values back and publishes them to MQTT.

So in your case, you just need to remove the `run_cycles: 15` and everything should work fine

oncleben31, 2018-06-08:
OK I understand. I've missunderstood the concept of cycle. From my point of view each measure from the dallas sensor was done in a cycle. When I put `run_cycles: 15` I was expecting to see 15 measures published on mqtt topic and then going to deep sleep.

closed 2018-06-08 by oncleben31