esphome/esphome #40: hassio addon 1.6.2 plataformio name resolution

codedmind, 2018-06-08:
Using hassio addon 1.6.2 after run the wizard for nodemmcu esp8266 12e i get the following error
esphomeyaml:
name: termoacumulador
platform: ESP8266
board: esp12e

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ArduinoOTA%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6a96761150>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=5)

OttoWinter, 2018-06-08:
So this is probably because platformio's servers are acting up again. They have done that a couple of times this week already :( (https://community.home-assistant.io/t/esphomelib-a-comprehensive-solution-for-using-esps-with-home-assistant/55349/2?u=ottowinter). Anyway, nothing I can do about that, since esphomeyaml is only possible with platformio's awesome build system. And to be fair, their server have pretty much had no downtime over the past 2 years.

Another potential issue could be your DNS settings for HassIO. That's probably not it, but if you want to check you could run `ping api.platformio.org` in the SSH add-on.

codedmind, 2018-06-08:
I run hassion in a linux virtual machine, from that machine ssh i can ping/nslookup api.platformio.org without issues.
Should i try inside esphomeyaml container?

OttoWinter, 2018-06-11:
@codedmind Hm, maybe would be good to check that too. I mean I don't really know all too much about your setup and networking in general, so I think I can't really help here

codedmind, 2018-06-11:
I managed to bypass the problem.
I don't know why, but my addons containers cannot resolve hostnames.
So i put api.platformio.org ip address in /etc/hosts and i'm able to compile the project.

I must add a new line to the nodemcu after the first update so ota update works then, because by name dns won't work either.

Thanks

mrand, 2018-07-12:
@codedmind 

I resolved this in two steps:

1. Created /etc/docker/daemon.json with the following:
`{ "dns": [ "172.17.0.1" ] }`
2. Created  /etc/NetworkManager/dnsmasq.d/docker-bridge.conf with the following:
'interface=docker0'

Then restart network manager (sudo service network-manager restart) and docker (sudo service docker restart, although hass-io doesn't auto restart for me, so easier to just reboot).

Hassfan, 2018-07-24:
@mrand where did you configure that - in the esphomeyaml container or the hassio container ?

I'm not having much luck trying to resolve this same issue :-(

mrand, 2018-07-24:
@Hassfan I did this outside of docker... i.e. the host - in my case, Ubuntu.

patraff, 2018-07-25:
I too have this issue. It has only happened recently as I have been playing with esp8266 and hassio for the past few months. Disclaimer - I am a tinkerer and have little formal knowledge of linux etc.
I have been able to compile on the esp8266 running espeasy in the past. Hassio at that stage was running on rasppi 3b+ (using daleh install). I now have set up hassio on a ubuntu server 18.04 NUC box and can't compile. I tried the fix by mrand but no success. I'll see if the same issue occurs when i fire up the pi again to check. I do all updates etc as they come available. Sorry - no assistance here

Hassfan, 2018-07-25:
thanks @mrand - I will give that a try.

@patraff - I had a quick look at the Hassio discord channel earlier and saw a suggestion there that docker and ubuntu 18.04 have DNS issues. I guess that is not helping with this.

I tested installing the esphomeyaml docker container on Ubuntu 17.04 and I am getting further with that, but now I have issues with the USB passthrough as I have this running in virtualbox!

If you haven't already guessed I am also learning this as I go and a linux novice!

codedmind, 2018-07-25:
I'm thinking about change the host to centos... but for now this post solve
me the issue with internet access

https://superuser.com/questions/1130898/no-internet-connection-inside-docker-containers

Last post

*1. Hardcode DNS server in docker daemon.json*

   -

   Edit /etc/docker/daemon.json

   {
       "dns": ["10.1.2.3", "8.8.8.8"]
   }

   -

   Restart the docker daemon for those changes to take effect:
   sudo systemctl restart docker
   -

   Now when you run/start a container, docker will populate /etc/resolv.conf
    with the values from daemon.json.

------------------------------

*2. Fix the hosts's /etc/resolv.conf*

*A. Ubuntu 16.04 and earlier*

   -

   For Ubuntu 16.04 and earlier, /etc/resolv.conf was dynamically generated
   by NetworkManager.
   -

   Comment out the line dns=dnsmasq (with a #) in
   /etc/NetworkManager/NetworkManager.conf
   -

   Restart the NetworkManager to regenerate /etc/resolv.conf :
   sudo systemctl restart network-manager
   -

   Verify on the host: cat /etc/resolv.conf

*B. Ubuntu 18.04 and later*

   -

   Ubuntu 18.04 changed to use systemd-resolved to generate /etc/resolv.conf
   <http://manpages.ubuntu.com/manpages/bionic/man8/systemd-resolved.service.8.html#contenttoc3>.
   Now by default it uses a local DNS cache 127.0.0.53. That will not work
   inside a container, so Docker will default to Google's 8.8.8.8 DNS server,
   which may break for people behind a firewall.
   -

   /etc/resolv.conf is actually a symlink (ls -l /etc/resolv.conf) which
   points to /run/systemd/resolve/stub-resolv.conf (127.0.0.53) by default
   in Ubuntu 18.04.
   -

   Just change the symlink to point to /run/systemd/resolve/resolv.conf,
   which lists the real DNS servers:
   sudo ln -sf /run/systemd/resolve/resolv.conf /etc/resolv.conf
   -

   Verify on the host: cat /etc/resolv.conf

Now you should have a valid /etc/resolv.conf on the host for docker to copy
into the containers.

Hassfan, 2018-07-25:
@codedmind - I think that is pretty much the same as suggested by @mrand above. I've just run through that and that seems to have solved this for me too. Thanks both for the help!

codedmind, 2018-07-25:
Correct, I only posted because the post refer diferrent solutions based on
Ubuntu versions

patraff, 2018-07-26:
Thanks for that
I did the modifications as set out and got the desired results of dns
servers as per the instructions. Would i need to reflash my esp8266? I ask
this because after the changes and a restart of hassio I still get the api
error from esphomeyaml compile.
Patrick

OttoWinter, 2018-11-27:
I believe Hass.io has fixed this issue in the latest few versions.

closed 2018-11-27 by OttoWinter