esphome/esphome #43: Hassio addon not working

ioudeveldhuis, 2018-06-08:
I have create 2 yaml files for my sonoff modules but when I press compile or validate buttons I get a terminated process message and nothing is done. I have not attached the module yet to flash because I want to migrate from Tasmota to this.

My yaml file looks like this:

esphomeyaml:
  name: Sonoff2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'XXXXXXXXX'
  password: 'XXXXXXXXXXX'

mqtt:
  broker: 'XXXX.XXXXXXXXXXXX.XXX'
  username: ''
  password: ''

# Enable logging
logger:

ota:
  password: 'XXXXXX'

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Sonoff Basic Button"
  - platform: status
    name: "Sonoff Basic Status"

switch:
  - platform: gpio
    name: "Sonoff Basic Relay"
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin: GPIO13
    inverted: True

light:
  - platform: monochromatic
    name: "Sonoff Basic Green LED"
    output: basic_green_led

tonertiffi, 2018-06-10:
I have create 1 yaml files for my sonoff modules but when i upload the sonoff.bin file (using a usb to serial flash programmer esptool.exe -vv -cb 115200 -cp com11 -ca 0x00000 -bz 1m -bm dout -cf sonoff.bin ,also logged into the web interface upload firmware says its ok )
Unfortunately nothing happens just 1 blink of the led and nothing.
the hardware is a sonoff basic 1 relay esp8266
the bin creates 100% validate is also 100%

any help would be a blessing tried so may thing already

config file in esphomeyaml

esphomeyaml:
  name: sonoff01
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'my router ssid'
  password: 'password of router'
  
  manual_ip:
    static_ip: 192.x.x.x
    gateway: 192.x.x.x
    subnet: 255.255.255.0  

mqtt:
  broker: 'server ip of hassio'
  username: 'my username'
  password: 'my password'

# Enable logging
logger:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "sonoff01 Basic Button"
  - platform: status
    name: "sonoff01 Basic Status"

switch:
  - platform: gpio
    name: "sonoff01 Basic Relay"
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "sonoff01 Basic Green LED"
    output: basic_green_led

OttoWinter, 2018-06-10:
@tonertiffi What specifically is not working? Does it not connect to WiFi/MQTT or does nothing show up in Home Assistant? The one short blink of the LED is telling me that esphomelib is at least starting up - maybe the WiFi/MQTT connection is failing?

The first step would be to view the esphomelib debug logs. Through the HassIO web interface, you can do this by clicking on the logs button (if you have it connected via USB, make sure you select `/dev/ttyUSB<something>` in the top navigation bar). Does something show up there?

Also, if you have the Arduino IDE installed or have a serial monitor, you can try starting a serial port with baud rate 115200. The logs should definitely help with debugging the issue.

tonertiffi, 2018-06-10:
Hi OttoWinter I have made progress still not completely there, debug .it logs into Mqtt I added the web server so I can log into it an do the basic turn on relay led on board still a no go, even from web page.Hassio nothing showing. Any suggestions would be much appreciated. Thanks in advance

OttoWinter, 2018-06-10:
@tonertiffi You can enable AP mode using the `ap:` option here: https://esphomelib.com/esphomeyaml/components/wifi.html

So do I understand this correctly?
 1. You can see the MQTT logs?
   * If not, that most probably means that the WiFi/MQTT ssid/server/password settings might be wrong.
    * If you're using `hassio.local` for the `mqtt.broker` option, consider using a static IP instead. Some routers do not like `hassio.local`
 2. The web server is not reachable, even though `web_server:` is in the configuration?
   * This would probably mean that the WiFi settings are not correct - or your router is somehow blocking the connection.
 3. Or alternatively, you cannot control *switches* through the web server.
   * This is a known bug in 1.6.2 (see https://github.com/OttoWinter/esphomelib/issues/68) - I've fixed it but for now you have to [use the latest dev version](https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version) to not have this bug
 4. The Home Assistant frontend is not showing the sensors/relays.
   * This could be, for example, be because [MQTT Discovery](https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version) is not explicitly enabled in the Home Assistant configuration.

tonertiffi, 2018-06-12:
> These are the last lines of the log.
>
> INFO [tornado.access] 304 GET /serial-ports (192.168.1.2) 5.57ms INFO [tornado.access] 304 GET / (192.168.1.2) 2.65ms INFO [tornado.access] 304 GET /static/materialize-stepper.min.css (192.168.1.2) 46.20ms INFO [tornado.access] 304 GET /static/materialize-stepper.min.js (192.168.1.2) 2.25ms INFO [tornado.access] 304 GET /serial-ports (192.168.1.2) 5.55ms INFO [tornado.access] 101 GET /validate (192.168.1.2) 2.05ms INFO [tornado.access] 304 GET /serial-ports (192.168.1.2) 5.58ms INFO [tornado.access] 304 GET /serial-ports (192.168.1.2) 5.55ms

Hi OttoWinter,

Thank you for your help,the suggestions, worked 100% got it working thank you very much.

If I could be so bold and ask you 1 more thing,
I have no idea how to do it,i would like to pull the relay in and led green to turn on at the same time, as this is a good visual indication that the relay is on. I do see them separate in my config file but how to combine them in 1.

Kind regards
Tonertiffi

OttoWinter, 2018-06-12:
Well that's certainly very **bold**, even a bit *italic* 😉

So the way I would do that is by using a template switch like this:

```yaml
switch:
  - platform: gpio
    # ...
    id: switch1
  - platform: gpio
    # ...
    id: switch2
  - platform: template
    name: "My Switch"
    turn_on_action:
      - switch.turn_on:
          id: switch1
      - switch.turn_on:
          id: switch2
    turn_off_action:
      - switch.turn_off:
          id: switch1
      - switch.turn_off:
          id: switch2
    optimistic: true
```

tonertiffi, 2018-06-13:
Hi OttoWinter,

Thank you very much I will be sure to try this.
Your help is very much apriciated

Kind regards
Tonertiffi

tonertiffi, 2018-08-29:
Good Day OttoWinter

Wonder if you could shed some light on my problem I am having,I have just received a Pi 3 model B+, an have a problem when i try an install esphomeyamal, i am using Hassio all that happened when i click the install button is, that it goes red, does not install or give me anything else, I did read the documentation put in the correct repository, unless there is a different one. It is running on 64bit.

Any help would be apriciated

Kind regards
Wayne

________________________________
From: hassio scott <hassio.scott@outlook.com>
Sent: Wednesday, June 13, 2018 9:43:23 AM
To: OttoWinter/esphomeyaml
Subject: Re: [OttoWinter/esphomeyaml] Hassio addon not working (#43)

Hi OttoWinter,

Thank you very much I will be sure to try this.
Your help is very much apriciated

Kind regards
Tonertiffi

OttoWinter, 2018-09-19:
>  Pi 3 model B+,

that's the problem. That specific RPi uses a specific CPU architecture (`aarch64`) for which the platformio project has not uploaded a compiler yet/the one that exists is very buggy. I'm also waiting for the day when such compiler is uploaded but until then esphomeyaml unfortunately can't support RPi 3 model B+.

OttoWinter, 2018-11-27:
Tracking the issue here https://github.com/OttoWinter/esphomeyaml/issues/81

closed 2018-11-27 by OttoWinter

tonertiffi, 2018-12-22:
Good day OttoWinter,

Thank you for taking the time to read this.
Please could I ask you if you could help me with some thing quickly, I am using your esphomelib,and loving it, thank you for the time you have spent on it, I actually went an bought a spare PI3 just to run Hassio and esphomelib as a development platform because PI 3 model b+ would not work.

I created the binary sensor window, its not a big thing if it cant do it but is there a way to change it to open and closed instead of the default on off please. Please bare in mind I know very little about code or programming, I know it has to do with mqtt messages.

I would very much appreciate your assistance.

Thanking you kindly
Wayne

Closed #43<https://github.com/OttoWinter/esphomeyaml/issues/43>.

Caswell1000, 2019-06-18:
Hello,
Managed to compile the firmware on hassio, there is no usb option to upload the firmware, I have restarted the add-on, but I still get the OTA option only, I also tried to upload the .bin file via the esphome flasher but it is also trying to connect via wifi as it gives a time-out, here's the begining of the code, which compiles, are there other ways to upload the bin file...

```name: sprinkler_system
  platform: ESP8266
  board: nodemcuv2
  includes:
    - shift_register_switch.h
  libraries:
    - ShiftRegister74HC595@1.2

wifi:
  ssid: "*********"
  password: "*******"
```