esphome/esphome #44: Easy control of RGB LED controler ?

vincegre, 2018-06-09:
Hi

I'm using these Wifi LED strips controler based on ESP:
[image]
It's fairly basic and use one GPIO for each color (R/G/B). Is it me or it's a little overcomplicated for basic control as I need to declare each color output like that:

```
output:
  - platform: gpio
    pin: 25
    id: 'color_red'
```
and then:

```
light:
  - platform: rgb
    name: "Living Room Lights"
    red: color_red
    green: color_green
    blue: color_blue
```
Isn't it possible to directly declare GPIO in light platform ?

Thanks

Vincèn

OttoWinter, 2018-06-09:
You need to use the `esp8266_pwm` output platform. The GPIO output platform can only produce **binary** signals whereas RGB lights require a floating point output.

See the note here: https://esphomelib.com/esphomeyaml/components/light/rgb.html

> The RGB light platform only works with float outputs that can output any light intensity percentage like the ESP32 LEDC or ESP8266 PWM components and does not work with output platforms like the GPIO Output.

So in your case it would be:

```yaml
output:
  - platform: esp8266_pwm
    pin: 25
    id: 'color_red'
 # Same for other outputs

light:
  - platform: rgb
    name: "Living Room Lights"
    red: color_red
    green: color_green
    blue: color_blue
```

vincegre, 2018-06-09:
Thanks so I did that config file for my controler:
```
esphomeyaml:
  name: sdb_light
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'myssid'
  password: 'mypass'

mqtt:
  broker: '192.168.1.8'
  username: 'myuser'
  password: 'mypass'

# Enable logging
logger:

ota:

output:
  - platform: esp8266_pwm
    pin: GPIO05
    id: 'color_red'
  - platform: esp8266_pwm
    pin: GPIO14
    id: 'color_green'
  - platform: esp8266_pwm
    pin: GPIO12
    id: 'color_blue'

light:
  - platform: rgb
    name: "Barre de LEDs"
    red: color_red
    green: color_green
    blue: color_blue
```
But once rebooted and connected to MQTT/HA, only a switch shows up in HA, no RGB control ???

OttoWinter, 2018-06-09:
Did you previously have a switch with the name "Barre de LEDs"? Because then you might need to restart Home Assistant.

vincegre, 2018-06-09:
@OttoWinter nope the "Barre de LEDs" is my new RGB controler so it appeared when I putted power on my newly flashed device in ha ! But only a switch appears, no RGB thing ! Previous automatic descovery devices have alwasy showed up automagicaddy in HA without reboot but rebooted it just in case and still the same ! Autodiscovery not working for RGB thing ?

OttoWinter, 2018-06-09:
Hmmm, I wonder if the controller you're using is running out of memory to send the discovery message. Do you see any `Publish failed` messages in the logs? Or anything else in the logs?

vincegre, 2018-06-10:
oki so it's working now fine in HA (out of blue that never goes on but I guess my LED controler has a default, the two other colors working !) ;)

Thanks for help

snizzleorg, 2018-08-07:
@vincegre  How did you flash esphomeyaml on the controller and where did you purchase that thing?

vincegre, 2018-08-07:
> How did you flash esphomeyaml on the controller and where did you purchase that thing?
@snizzleorg  Flashing process is the usual one for ESP, it's just a pain in the ass to connect to it as there is no connector for it on board and board is pretty tiny and dense ! So you have to directly solder the 3 wires for flashing (TX, RX, GND) on the ESP and same for flash button....
For purchase you'll find these very easily on Aliexpress either alone or with some RGB strips, and with/without power supply. For example: https://www.aliexpress.com/item/WiFi-Control-RGB-LED-Strip-5050-Set-Home-Decoration-Neon-Light-with-Mini-WiFi-RGB-Controller/32716841823.html?spm=a2g0s.9042311.0.0.27424c4daJ9zZ3 Be careful it exists too with RF/IR remote (probably similar electronic inside but not checked !!).

snizzleorg, 2018-08-08:
@vincegre Thanks!. I was afraid that the flashing involved soldering... that kind of a bummer. thanks anyways!

vincegre, 2018-08-08:
> Thanks!. I was afraid that the flashing involved soldering... that kind of a bummer. thanks anyways!

Yep sorry it's definitively not a Sonoff hackable designed product :-1:  but you can also do RGB control easily using an ESP ;)

snizzleorg, 2018-08-08:
yeah but then I need a PCB and some housing. Maybe easier and faster to buy something like a hue strip.

vincegre, 2018-08-08:
> Thanks!. I was afraid that the flashing involved soldering... that kind of a bummer. thanks anyways!

Good point but still will need a little of soldering :+1:

vincegre, 2018-11-30:
> This is one of the devices listed here, right? https://esphomelib.com/esphomeyaml/devices/sonoff.html#arilux-lc10-magic-home

Not sure but will give it a try and see results ;)