esphome/esphome #45: Web server makes ESP crash on Sonoff basic !

vincegre, 2018-06-09:
Just using latest version of tool. I have done following configuration file for my Sonoff basic:
```
esphomeyaml:
  name: sdb_light
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'myssid'
  password: 'mywifipass'

mqtt:
  broker: '192.168.1.8'
  username: 'mymqttlogin'
  password: 'mymqttpass'

# Enable logging
logger:

ota:

switch:
  - platform: gpio
    name: "SDB Lumière"
    pin: GPIO12
```

Configuration likes that works perfectly ! If I add the web component, as soon as I play one or two times with the on/off button in web interface the ESP crashes !

Vincèn

OttoWinter, 2018-06-09:
Probably related to https://github.com/OttoWinter/esphomelib/issues/68 - I will have a look at it later today

vincegre, 2018-06-09:
Oups yep you are right ;) missed his post ! so will close this ticket and follow the other one !

closed 2018-06-09 by vincegre