esphome/esphome #46: Bug in OTA update process !

vincegre, 2018-06-09:
Using the OTA update with command line argument to indicate specific IP of my ESP, it uploads successfully the firmware file in OTA but then turns in serial mode for logs instead of doing it through IP (don't know if it's possible) or then close the process and that's it ;)

```
12:53:25 [DEBUG]: Options: {'esp_ip': '192.168.1.152', 'host_port': 53445, 'image': '.pioenvs/led_strips/firmware.bin', 'host_ip': '0.0.0.0', 'auth': '', 'esp_port': 8266, 'spiffs': False, 'debug': True, 'progress': True}
12:53:25 [INFO]: Starting on 0.0.0.0:53445
12:53:25 [INFO]: Upload size: 384880
Sending invitation to 192.168.1.152
12:53:25 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

12:53:38 [INFO]: Waiting for result...
12:53:38 [INFO]: Result: OK
================================================================ [SUCCESS] Took 15.16 seconds ================================================================
INFO [esphomeyaml.__main__] Successfully uploaded program.
INFO [esphomeyaml.__main__] Starting log output from 192.168.1.152 with baud rate 115200
Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 463, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 453, in run_esphomeyaml
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 325, in command_run
    return show_logs(config, args, port, escape=args.escape)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 218, in show_logs
    run_miniterm(config, port, escape=escape)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 103, in run_miniterm
    with serial.Serial(port, baudrate=baud_rate) as ser:
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port 192.168.1.152: [Errno 2] No such file or directory: '192.168.1.152'

```

OttoWinter, 2018-06-09:
Already fixed here #42 😉 See https://esphomelib.com/esphomeyaml/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version

vincegre, 2018-06-09:
Will wait it's pushed in mainstream, prefers to avoid using bleeding edge code :+1:

closed 2018-06-09 by vincegre