esphome/esphome #47: "no known conversion" error with IR raw transmitter

r-jordan, 2018-06-09:
I want to make an IR transmitter with the following yaml code:
<pre>
switch:
  - platform: ir_transmitter
    name: "ON/OFF"          
    raw:          
      carrier_frequency: 38kHz
      data:                   
        - 1300
        - -300
        - 1300
        - -300
        - 300 
        - -1300
        - 300  
        - -1300
        - 300  
</pre>

It's translated into the following C code by esphomeyaml:

<pre>
  auto *switch__switch = App.register_component(switch__irtransmittercomponent->create_transmitter("ON/OFF", switch_::ir::SendData::from_raw({
          1300,
          -300,
          1300,
          -300,
          300,
          -1300,
          300,
          -1300,
          300,
  }, 38000)));
</pre>

But I get this error at compile time:

<pre>
src/main.cpp: In function 'void setup()':
src/main.cpp:28:11: error: no matching function for call to 'esphomelib::switch_::ir::SendData::from_raw(<brace-enclosed initializer list>, int)'
}, 38000)));
^
src/main.cpp:28:11: note: candidate is:
In file included from .piolibdeps/esphomelib/src/esphomelib/application.h:63:0,
from src/main.cpp:2:
.piolibdeps/esphomelib/src/esphomelib/switch_/ir_transmitter_component.h:108:19: note: static esphomelib::switch_::ir::SendData esphomelib::switch_::ir::SendData::from_raw(std::vector<int>&, uint32_t)
static SendData from_raw(std::vector<int> &raw_data, uint32_t carrier_frequency);
^
.piolibdeps/esphomelib/src/esphomelib/switch_/ir_transmitter_component.h:108:19: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'std::vector<int>&'
*** [.pioenvs/device1/src/main.cpp.o] Error 1
</pre>

OttoWinter, 2018-06-09:
Yeah... noticed that too while rewriting the IR component. In the new IR component (or now it's called remote component) code that is fixed, but will only be a part of 1.7.0.

The problem is that there's a small `&` symbol that shouldn't be there on this line:

https://github.com/OttoWinter/esphomelib/blob/99965ed73195e0797c27123faf642b6b1a41ac1a/src/esphomelib/switch_/ir_transmitter_component.h#L108

and this one:

https://github.com/OttoWinter/esphomelib/blob/master/src/esphomelib/switch_/ir_transmitter_component.cpp#L43

If you know how to edit that file manually (should be somewhere like `<NODE_NAME>/.piolibdeps/<SOME_FOLDER>/esphomelib/switch_/ir_transmitter{.h,.cpp}`, you can remove the `&` manually. If not, I'm sorry you're going to have to wait for the 1.7.0 update 😥

r-jordan, 2018-06-10:
Yes, I will edit them by hand.

I've just began playing with esphomelib today and it's a great tool! Very easy and very powerfull. Thank you!

r-jordan, 2018-06-10:
I managed to delete the extra '&' symbols and it compiles now. But I haven't got any IR signal from the IR diodes. I used a simple oscilloscope and I don't see any signal on the output pin of the ESP8266.
I'm using the same circuit with other software and it works good.
Any else issue with the code?

OttoWinter, 2018-06-11:
Puh I don't know, I'm currently working on the update code and honestly don't intend to try to diagnose the old one, sorry

closed 2018-06-11 by r-jordan