esphome/esphome #50: Support for wind sensor (Eltako WS)

hanzoh, 2018-06-11:
While trying to build a weather station with several sensors,
I found this one to be a worthy addition: https://www.amazon.de/Eltako-Windsensor-WS/dp/B0018LBFG8

There is also a ready .ino file in this repo: https://github.com/sbiermann/esp8266-anemometer/blob/master/esp8266-anemometer.ino

Would this be possible to add to esphomelib/esphomeyaml?

OttoWinter, 2018-06-11:
Already possible with the pulse counter sensor: https://esphomelib.com/esphomeyaml/components/sensor/pulse_counter.html (which will even be backported to the ESP8266 in 1.7.0)

The only thing that would need to be done is a conversion from the pulse count to the wind speed. In the code I see this formula that should work: `wind = 1.761 / (1 + rps) + 3.013 * rps;`. See the TEMT6000 example for inspiration.

A formula could for example be:
```yaml
sensor:
  - platform: pulse_counter
    pin: GPIOx
    # ...
    filters:
      - lambda: >-
          return 1.761 / (1 + x) + 3.013 * x;
```

OttoWinter, 2018-06-12:
Yes, definitely. I myself am using a similar wind sensor and a single turn can create many hundred pulses without any filtering.