esphome/esphome #51: [Feature Request] nRF24L01 support

brandond, 2018-06-11:
I have need for some small battery-powered distributed sensors. Ideally this would be a good use for BLE or Z-Wave, but BLE support has a ton of overhead and Z-Wave is proprietary (can only be used with closed-source home gateway devices and cloud service providers).

What I'm probably going to end up doing is setting up a few devices with STM32s that will spend most of their time in suspend, occasionally waking to make a sensor reading and send it off via a [nRF24L01](https://platformio.org/lib/show/433/RF24) radio if conditions are met. On the ESP side, it would just be a matter of wiring up the RF24 SPI/Enable/IRQ pins to GPIOs. 

 The chip is limited in hardware to 32 byte message payloads. Perhaps something like this:
```c
struct payload {
  unit8_t type;
  char unit[3];
  char value[14];
  char name[14];
};
```

Ideally, esphomelib would not need to define a list of sensors ahead of time, but simply reformat and pass the messages through to MQTT.

OttoWinter, 2018-06-12:
So in principle, this would be awesome. But for one part I'm so certain:

> Ideally, esphomelib would not need to define a list of sensors ahead of time, but simply reformat and pass the messages through to MQTT.

I mean I do understand the usability aspect of that, but the backend sensor should not know about the front-end (here MQTT). esphomelib is purposefully not built in a way to make it depend on MQTT and in the future it might also support other frontends. For example, one such frontend is already the web server.

One way this could still not make usability too bad I hope is with how I recently did it with the `remote_receiver`. The remote receiver can listen on any pin for IR/433MHz/similar for any sequence of encoded data. For example, it can listen to the IR codes sent by a TV remote. Of course all these codes have to be learned first, that's why there are "dumpers" that just dump any received IR code in the logs together with their decoded values. The user can then just copy+paste this into their configuration and they're already done.

This way, the backend can expose individual binary sensors without ever knowing how the frontend handles them. In this case with these packets you have there, the same thing could be done with `sensors`, with the name identifying them or however the protocol works.

OttoWinter, 2018-06-14:
*Theoretically* it's possible to register them dynamically, you would only need to handle their setup lifecycle (with `setup_()`) manually.

Removing them dynamically too would probably be a bit more difficult, as there are many inter-object dependencies like callbacks that could break if a component on the heap is manually `delete`d.

But I would say a big drawback of such a solution would be that these dynamically created sensors would pretty much be inaccessible from automations and triggers.