esphome/esphome #53: Proposals for a logo

messismore, 2018-06-13:
Hi!

Inspired by this [Twitter thread](https://twitter.com/TheHookUp1/status/1005860946278051840) I have done some rough sketches for what I think would make a nice logo, or, rather a starting point. They are pretty straightforward in their metaphor (a stylised ESP8266 as mounted on the NodeMCU in varying degrees of abstraction + a house) but I think the simplicity works rather well.
[image]

I really liked @OttoWinter's idea to work with the chip's antenna, as it is quite a distinct feature, however I am also drawn to no.1 visual simplicity. Maybe someone will be inspired to a eureka moment by one of these.

Note that I kept the name esphome*lib* for now, as @OttoWinter is still thinking about the name.

[image]
[esphomelib logo proposal.pdf](https://github.com/OttoWinter/esphomeyaml/files/2099753/esphomelib.logo.proposal.pdf)

Cheers

carlosmgr, 2018-06-15:
I Like the 5, it's simple

brandond, 2018-06-15:
I like 3 shows the recognizable antenna, plus I like the home being 'in' the chip.

Landrash, 2018-06-15:
I give a vote for 5 and 3.

I suggest @OttoWinter adopt the one he likes but having svg/png files available in the documentation repository would be rather awesome.

@messismore Have you considered nr 3 with a hollow house instead?

messismore, 2018-06-15:
>  I would be happy to incorporate the design into the website and project if you're okay with it

Absolutely! Should you decide to adopt one of them, I'd go over it some (harmonise it with the font's dimensions) and issue a pull request with a Svg and Png.

@Landrash I did, but I think the solid house provides some nice visual weight next to the wordmark:
[image]

jackjohnsonuk, 2018-06-19:
Number 5 👍

Landrash, 2018-06-20:
@messismore I agree with you on that it looks beter with the filled house.

mefizto, 2018-06-25:
3

zmrow, 2018-07-03:
My vote is #3

cgtobi, 2018-08-16:
My votes for 5 and 3, in that order. :) Great logos. They'd be recognisable on their own without the text quite nicely.

justinmartin, 2018-08-21:
Number 3 is tops for me. Followed closely by 5.

OttoWinter, 2018-09-19:
@messismore Sorry for the waiting. I've had to concentrate on other topics for a while (work, ioi, university, ...). I would be really happy to finally see esphomelib with a logo and I've spent the past time contemplating a lot about the marvelous logos 5 & 3 and in this thread they both seem to share about the equal amount of fans :)

For the next release I would really like to have the new logo ready. And.... I have decided that I want to go for number 3 :) When you have time, could you do the final cleanups you wanted to do and create a PR with an image file? Anyway, if you don't have time right now (which is 💯 ok), I mean I can also just use a cropped picture from above and maybe update it later.

Thank you again very much. Your work is excellent and very much appreciated!