esphome/esphome #54: install in python virtual environment?

oldfart101, 2018-06-13:
Home Assistant running in python virtual enviroment on Hassbian on PI3, Python 3.5.3
how to install??

brandond, 2018-06-13:
Same as any other Python module?
```bash
source /path/to/venv/bin/activate
pip install esphomeyaml
```

Some folks have reported that the espressif toolchain (compiler, etc) tends to crash on arm, though.

oldfart101, 2018-06-14:
pip install esphomeyaml
Collecting esphomeyaml
  Could not find a version that satisfies the requirement esphomeyaml (from versions: )
No matching distribution found for esphomeyaml

oldfart101, 2018-06-14:
Yes, I had read the requirements for esphomeyaml, which is why I put Python 3.5.3 in the first post. So there is no way to install in my Virtual environment, since it is running Python3, can it be run outside the environment and connect to HA?

brandond, 2018-06-14:
You'd need to create a separate virtualenv using Python 2.7. esphomeyaml usually interacts with HA via MQTT (over the network), so there's no reason they would need to be installed in the same virtualenv or even on the same host.