esphome/esphome #55: [Feature Request] make rest calls

snizzleorg, 2018-06-15:
Would be nice if one could make REST calls to directly send commands to devices which accept those (e.g. hue bridge).

OttoWinter, 2018-06-19:
Sorry but no.

I don't really understand why you would do it this way. I mean you can do REST API calls from Home Assistant too (for example using the [rest](https://www.home-assistant.io/components/switch.rest/) [components](https://www.home-assistant.io/components/rest_command/)), and I guarantee you those will work much better than esphomelib can achieve.

To clarify, esphomelib's goal is *not* to replace your home automation system by itself, it's meant to work well together with your exisiting home automation system like Home Assistant and integrate your ESP into it. esphomelib's goal is for example to control devices that are directly connected to it, like RGB led strips. Anything else, like controlling Hue hubs or reading weather data from an external service like WUnderground should be done by other solutions like Home Assistant.

Also: Adding a full HTTP stack on a tiny microcontroller like an ESP is incredibly inefficient, and the HTTP libraries that already exist for the ESP surely have many bugs. You're probably much better off using a solution on a powerful host like an RPi that has well-tested HTTP libraries like an RPi with Home Assistant.

snizzleorg, 2018-06-22:
Of course I can do rest calls from Home Assistant and this is working beautifully. The problem however is that I have to use an automation to do that. And this is rather slow. So for example I integrated a rotary_encoder via esphomeyaml which is a sensor in home-assistant. Then I use an automation to set the brightness depending on the sensor. 
And this is where it becomes unusable. The reaction times can be on the order of 1-2 seconds. While a normal person would expect the dimmer to react instantly. 

My hope was that when using a rest call to directly set the brightness from esphomeyaml the reaction time would be improved...

But I totally understand the argument about the scope...