esphome/esphome #56: [Feature Request] Input validation

mrand, 2018-06-15:
When a user enters a device name with upper-case characters, it fails to validate or compile.  But since the plugin doesn't appear to have a way to go back and edit the file, they will have to go to a configuration editor or jump over to command line to fix it.

The failure could be avoided in the first place if ESPhomeYAML did a first pass simple validation on the input field in initial config, or else did an automatic conversion to lower-case.

balk77, 2018-06-19:
Same for spaces in the name

OttoWinter, 2018-06-19:
I understand the problem and real input validation would be great. However I have no idea how to do it with the material stepper library I used; probably it's somehow possible with the jquery validation plugin it uses, but I don't know how.

If someone knows how to do it, then a PR to fix this behavior would be very much appreciated. I personally just don't think this is that big of a problem and don't want to have to debug weird javascript errors for many more hours (I'm a quite inexperienced *web* developer).

To address the issue though I would propose to do some "social" hacking by changing the

> It may only contain the characters `a-z`, `0-9` and `_`

message on that screen to a

> Names must be **lowercase** and **must not contain spaces** (allowed characters: `a-z`, `0-9` and `_`)

balk77, 2018-06-19:
Excellent resolution :)