esphome/esphome #57: remote_recever won't compile : error: 'esp_err_to_name' was not declared in this scope

thomashermine, 2018-06-15:
Hi,

I'm trying to add the remote_receiver to sniff some 433mhz remote control signals.
But, I soon as I add the remove_receiver component, esphomeyaml won't build.

Here is the errors I got:
```
<command-line>:0:16: warning: ISO C++11 requires whitespace after the macro name
In file included from .piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:10:0:
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp: In member function 'virtual void esphomelib::remote::RemoteReceiverComponent::setup()':
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:155:77: error: 'esp_err_to_name' was not declared in this scope
ESP_LOGE(TAG, "Configuring RMT remote failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/log.h:124:130: note: in definition of macro 'esph_log_e'
#define esph_log_e(tag, format, ...) esp_log_printf_(ESPHOMELIB_LOG_LEVEL_ERROR, tag, ESPHOMELIB_LOG_FORMAT(tag, E, format), ##__VA_ARGS__)
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:155:5: note: in expansion of macro 'ESP_LOGE'
ESP_LOGE(TAG, "Configuring RMT remote failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:162:76: error: 'esp_err_to_name' was not declared in this scope
ESP_LOGE(TAG, "Installing RMT driver failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/log.h:124:130: note: in definition of macro 'esph_log_e'
#define esph_log_e(tag, format, ...) esp_log_printf_(ESPHOMELIB_LOG_LEVEL_ERROR, tag, ESPHOMELIB_LOG_FORMAT(tag, E, format), ##__VA_ARGS__)
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:162:5: note: in expansion of macro 'ESP_LOGE'
ESP_LOGE(TAG, "Installing RMT driver failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:168:81: error: 'esp_err_to_name' was not declared in this scope
ESP_LOGE(TAG, "Getting RMT ringbuf handle failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/log.h:124:130: note: in definition of macro 'esph_log_e'
#define esph_log_e(tag, format, ...) esp_log_printf_(ESPHOMELIB_LOG_LEVEL_ERROR, tag, ESPHOMELIB_LOG_FORMAT(tag, E, format), ##__VA_ARGS__)
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:168:5: note: in expansion of macro 'ESP_LOGE'
ESP_LOGE(TAG, "Getting RMT ringbuf handle failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:174:81: error: 'esp_err_to_name' was not declared in this scope
ESP_LOGE(TAG, "Starting RMT for receiving failed: %s", esp_err_to_name(error));
^
.piolibdeps/esphomelib/src/esphomelib/log.h:124:130: note: in definition of macro 'esph_log_e'
#define esph_log_e(tag, format, ...) esp_log_printf_(ESPHOMELIB_LOG_LEVEL_ERROR, tag, ESPHOMELIB_LOG_FORMAT(tag, E, format), ##__VA_ARGS__)
^
.piolibdeps/esphomelib/src/esphomelib/remote/remote_protocol.cpp:174:5: note: in expansion of macro 'ESP_LOGE'
ESP_LOGE(TAG, "Starting RMT for receiving failed: %s", esp_err_to_name(error));
^
*** [.pioenvs/remoteliving/libf8d/esphomelib/esphomelib/remote/remote_protocol.cpp.o] Error 1
```

And here is my configuration file:

```
esphomeyaml:
  name: remoteliving
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: '<redacted>'
  password: <redacted>'

mqtt:
  broker: '<redacted>'
  username: '<redacted>'
  password: '<redacted>'

logger:

ota:
remote_receiver:
  pin: GPIO32
  dump: all
```

I already tried: 
- Uninstall/Reinstall esphomeyaml with pip
- Delete this node folder (./remoteliving)

What else could I try? Thanks for your help!
(And this greeeaaat tool!)

OttoWinter, 2018-06-15:
Ok, I think this is due to an old arduino version (at least my version includes `esp_err.h` where `esp_err_to_name` sits by default.

I just pushed a small update which includes that in the log file, could you try if it works? (Read the FAQ for using the master branch)

thomashermine, 2018-06-15:
Running 
`platformio update`

fixed the issue. I should have read the FAQ... my bad, sorry!

Sadly, I had no chance to check if your update to add a more precise error to the log was working before :-/

closed 2018-06-15 by thomashermine