esphome/esphome #60: Bme280 error reporting after upgrade from 1.6.2 to 1.7

alesurli, 2018-06-15:
After i upgraded to 1.7 my bme280 starts to report incorrect value, ad ex:
[sensor.bme280:193]: Got temperature=0.7°C pressure=258601.6hPa humidity=14.3%

reverting to 1.6.2 resolve the issue:

[sensor.bme280:193]: Got temperature=24.7°C pressure=1012.0 hPa humidity=64.3%

exxamalte, 2018-06-16:
I have a similar issue with a BME280 and a BH1750 (200lx at startup, then always 0lx), both connected via I2C. Issue started when migrating from 1.6.2 to 1.7.
```
[D][sensor.bme280:193]: Got temperature=0.5°C pressure=319545.9hPa humidity=16.3%
[D][sensor.bh1750:091]: 'Luminance': Got illuminance=0.0lx
```

exxamalte, 2018-06-17:
I'm working on a fix, and will test it tonight.

exxamalte, 2018-06-18:
I can now confirm that this issue was caused by a typo in the code reading data from the I2C bus, so this issue could affect other I2C connected sensors, too.
PR https://github.com/OttoWinter/esphomelib/pull/98 opened for this fix in the esphomelib project.

If you can't wait, feel free to use the following configuration for your sensors in the meantime:
```
esphomeyaml:
  library_uri: 'https://github.com/exxamalte/esphomelib.git#i2c-fix'
```

alesurli, 2018-06-18:
thanks! i'll test ASAP.

Amir974, 2018-07-29:
I am using MPU6050 with I2C - the goal is to get data from the Gyro 
Prior to using this library line - I could not get any data from the device on the log
After using this library URL - I manage to see updated on the log but on the MQTT I see one update as the device comes online - but then it seems static (On HomeAssistant - the sensors data does not change although the log shows different values on the various parameters....)

I don't know if the data is accurate but at least I see something on the log.

now the question is how I can see this movement reflected on HA sensors?