esphome/esphome #62: [Feature Request] Remember MQTT Broker Details and Wifi Details

jackjohnsonuk, 2018-06-19:
Have an option in the web UI to remember both your MQTT Broker details and the details for the Wifi, I think this could get pretty repetitive if you have a whole bunch of esp devices to set up. Perhaps even if we were able to give them a "default value" that might be suitable? Thanks

brandond, 2018-06-20:
You could always just do one and then copy the YAML file and change the name.

jackjohnsonuk, 2018-06-20:
Then the whole interface for adding devices is completely unused.  People use multiple ESP chips, manufacturers etc and they will change on every device.. I'd say it's unusual for people to change MQTT broker or wifi though.
But I agreed, simple enough to do, just seems like something that could also be baked in one day.  Trying to think of any way to make the product more end-user-friendly. 
Thanks for the suggestion though

OttoWinter, 2018-06-22:
I kind of agree with @brandond it's much easier anyway to just copy the configuration to a new file.

For example, I have 6 ESPs that are identical hardware-wise and so I just copy the configuration over each time I decide to build a new one.

For "saving" the WiFi & MQTT details: The nice thing about the dashboard interface is that it's currently 100% stateless. There's no file that needs to be loaded for it to start up and it doesn't save any dashboard-related files either.

A while ago there was a suggestion for automatically populating the WiFi field on HassIO using data stored by the host. Plus on HassIO, we can probably even use the Hass API to get the MQTT credentials. But honestly, that would be a lot of work for what seems like not too much added UX. Maybe at some point but not now.

commit 7264e3a4 (2018-06-06): Fix ultrasonic typo Fixes #62