esphome/esphome #63: Switch state in conditions ?

vincegre, 2018-06-21:
Hi

Do you plan to be able to trigger some automations on some input.switch ? I have sonoff devices and I'd like to program them so one of the LEDs on it reflects the state of the relay ;)

Thanks

brandond, 2018-06-21:
I think the current recommended way to do this is to:
* Set up the LED as a binary light, but don't give it a name (to prevent it from showing up in HASS)
* Set up the relay as a switch, also without a name
* Set up a template switch, configure it to turn the LED and relay on and off together

OttoWinter, 2018-06-22:
Yes, as @brandond said: currently that's only possible with a template switch. Kind of like this:

```yaml
# Pseudocode!
switch:
  - platform: gpio
    id: led
    pin: D4
  - platform: gpio
    id: relay
    pin: D5
  - platform: template
    name: My Switch
    turn_on_action:
        - switch.turn_on:
            id: led
        - switch.turn_on:
            id: relay
    turn_off_action:
        - switch.turn_off:
            id: led
        - switch.turn_off:
            id: relay
```

As that's IMO pretty easy, I'm not sure I should add native support for it (would just end up cluttering docs). But for the "triggering on a switch", I guess that would be possible, might take a look at that once I get some time.

TheJulianJES, 2018-07-25:
Yeah, I hoped something like this would work:
```
binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - switch.toggle:
            id: relay

switch:
  - platform: gpio
    id: relay
    name: "Sonoff3"
    pin: GPIO12
    turn_on_action:
      then:
        - light.turn_on:
          id: led_light
    turn_off_action:
      then:
        - light.turn_off:
          id: led_light

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    id: led_light
    output: basic_green_led
```
The template switch method seems to work though. If implementing the turn_on//-off actions in the (gpio) switch (or switch core?) is possible, then that would be wonderful!

OttoWinter, 2018-07-28:
@TheJulianJES Why not something like this?

```yaml
binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - switch.toggle:
            id: relay
        - light.toggle:
            id: led_light
```

TheJulianJES, 2018-07-28:
Yeah sorry, I should have updated that comment. I already got that solution on discord although I previously tried something that worked but spammed mqtt, even after a slimmed down version (which doesn't work but shouldn't spam mqtt?): https://i.imgur.com/0jE3xCt.png // https://i.imgur.com/0DlpKZH.png (Maybe a bug?) https://hastebin.com/zocicupata.m was another solution I came up with but that might have been a bit overcomplicated. So, then somebody sent an option like yours on Discord (makes sense, but I didn't think of that for some reason).

vincegre, 2018-07-30:
> binary_sensor:
>   - platform: gpio
>     id: button
>     pin:
>       number: GPIO0
>       mode: INPUT_PULLUP
>       inverted: True
>     on_press:
>       then:
>         - switch.toggle:
>             id: relay
>         - light.toggle:
>             id: led_light

Nice but not fault proof as if led_light has been toggled for an other reason light will be then in opposite state ! Would be better and more fault proof to put light in same status as relay ;)

michamichamicha, 2018-07-30:
I tried using the led as power_supply option but that does not work for me. Not sure why but could also be an option?

vincegre, 2018-07-30:
> I tried using the led as power_supply option but that does not work for me. Not sure why but could also be an option?

What are you trying to do exactly ?

michamichamicha, 2018-07-30:
I'm trying to do the exact same thing as you are :) Making the led reflect the state of the relay.
I tried using the power_supply option, so the board would enable the led as if it were the power supply for the relay. This would also allow it to shut it down after the relay has been disabled.

```
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    id: bogus_2
    on_press:
      then:
        - light.toggle:
            id: living_lights_front_2

power_supply:
  - id: basic_green_led
    pin:
      number: GPIO13
      inverted: True
    keep_on_time: 0.5s

output:
  - platform: gpio
    pin:
      number: GPIO12
    id: 'generic_out'
    power_supply: basic_green_led

light:
  - platform: binary
    name: "Living room lights front 2"
    output: generic_out
    id: 'living_lights_front_2'
```

But for some reason it's not working. Still, this is a workaround and it would be nice if we can make multiple outputs work in sync. Maybe this can be done by setting the correct MQTT payload topic? Not sure if it's available for the `light` component.