esphome/esphome #66: change icon

mefizto, 2018-06-25:
how can i change the icon, i try:

`light:
  - platform: rgb
    name: "Luz de Escritorio"
    icon: "mdi: track-light"
    red: output_component1
    green: output_component2
    blue: output_component3`

 but it doesnt work, it cant validate

brandond, 2018-06-25:
Did you try it without a space in the icon value?
```yaml
light:
  - platform: rgb
    icon: "mdi:track-light"
```
```

OttoWinter, 2018-06-25:
Manually setting an icon is not possible with lights. You can always override it in the `customize` section in home assistant.

This has two reasons:
 1. Home Assistant's MQTT JSON lights do not support it and I doubt support for it will be added.
 2. It keeps the configuration clean(er): I don't want to add configuration options for every single thing that *could* be done, especially if it's very easy to do in Hass too.

mefizto, 2018-06-26:
thank you very much

closed 2018-06-26 by mefizto