esphome/esphome #67: [Suggestion / Feature Request] Configuration Templating

sh0rez, 2018-06-25:
Hey folks!
esphomeyaml is a great way of managing the functionality of a single device, but quickly fails when trying to manage a bigger fleet of devices.

# TL;DR
Split the configuration into templates and instances, generate the final code for each instance

# The Problem
As long as the number of ESP's stays quite low, it's no problem to duplicate the `.yaml` for every single ESP.
But as I plan to deploy in massive scale (30+ hardware equal devices), this is no option because of the following reasons:

- Most of these duplicated files would contain the same .. actually only the service names are going to change.
- As soon as I need to change anything on all nodes of the fleet, I would need to edit ALL files, which is not reasonable in any way.

# The Solution
To solve this, I suggest a templating system for the configuration files. The idea is stolen from proved templating systems used in web-development. (e.g. Handlebars, mustache, etc.)

The idea is the following:
The configurations gets split into a template, and into instances. The template describes, how all devices together are and the instances contain specific data about a single device.

Instead of creating a single `.yaml`-file for the configuration, it gets a folder. This could look as follows:

<hr>

*Configuration of hardware type "wifi_socket":*
**Folder name:** `.../home/wifi_socket/`. where `home` is the root of all hardware types, `wifi_socket` an example hardware type

**Contents:**
- `wifi_socket.yaml`: Basically a regular esphomeyaml-config, the same way as it is right now, but on steroides: Everything that is device specific, gets replaced by `{{ placeholder }}`. For example:

```yaml
# kitchensocket1.yaml
esphomeyaml:
  name: "Kitchen Socket #1"
```
gets:
```yaml
# wifi_socket.yaml
esphomeyaml:
  name: "{{ name }}"
```

At this point, the template is a general state, describing the whole fleet at once. To create the code for the specific instances, there are seperate `.yaml`-files:

```bash
/wifi_socket.yaml
/instances/
          /kitchensocket1.yaml
          /kitchensocket2.yaml
          /tvsocket.yaml
          /... .yaml
```

Each of these files contain simple key-value mappings, of the placeholders of the template:
```yaml
# kitchensocket1.yaml
name: "Kitchen Socket #1"
# (...)
```

Finally, for every instance, the specific values get merged into the final .yaml-file in memory and compiled to the code folder, the same way, esphomeyaml already does, just into it's own folder:

```bash
/wifi_socket.yaml
/instances/
          /kitchensocket1.yaml
          /kitchensocket2.yaml
          /tvsocket.yaml
/dist
          /kitchensocket1/
                         /lib
                         /src
                         /platformio.ini
          /kitchensocket2/
          /tvsocket/
```

<hr>

# Great outline .. but why is this useful?
This system has several advantages:
1. Most important: Hardware related stuff can be changed in a single file for ALL boards of the same type, while the specific board gets its specific data in it's dedicated configuration
2. Now one can edit all boards in a single file, reflecting the changes into every instance with no need to do 30+ equal changes.

# Final words
I really hope at least one person understands my idea lol .. or at least read until this point.
The implementation could be put into the esphomeyaml python app, or wrapped around it using a new application, or esphomeyaml could even be changed into an API, providing the ability to script on top of it.

brandond, 2018-06-26:
Templating would be interesting. Jinja2 is frequently used with YAML, but I'm not sure what exactly I'd want to fill in with it. Most of the common use cases I can think of (standard wifi, mqtt, etc) can already be accomplished using includes:

https://www.home-assistant.io/docs/configuration/splitting_configuration/

OttoWinter, 2018-06-26:
Ah sorry, my e-mail github notifications are currently not really working. Anyway, I agree this is something that should be improved, but I think template engines (like jinja2) are not the way to go.

While templating engines can accomplish a ton of customization ...

 * ... very few work well together with the language they're templating. I once tried to template my Hass YAML files with a custom preprocessor, but it's quite hard to get all the spaces and so on right.
 * ... it's another layer of complexity. Users would need to learn YAML, C++ (for lambdas) *and* a templating engine.
 * ... it can lead to situations where something get's templated even though you didn't want to do so. For example if I have `{{ }}` in my configuration somewhere it will be picked up by the preprocessor.

I think an adequate solution, like @brandond wrote, is using the existing YAML configuration tags like `!include` and YAML anchors. For example, you could have a `common.yaml` like this:

```yaml
wifi:
  ssid: ...
  password: ...
sensor:
  - platform: dht
    pin: D0
    # ...
binary_sensor:
  - platform: gpio
    # ...
```

and a per-device configuration like this:

```yaml
esphomeyaml:
  name: ...

# ...

.common: !include &common common.yaml
<<: *common
```

The `!include` tag will the `common.yaml` file to be read in into a `.common` section of your configuration with the YAML anchor `common`. And the `<<: *common` line is doing the magic of copying the common section into the per-device config

Note that esphomeyaml currently has a bug that prevents you from doing the above, but I've already fixed that locally.

If someone still needs a pre-processor like jinja2, they can just have a small script execute before esphomeyaml that does the pre-processing.

--------

> esphomeyaml could even be changed into an API, providing the ability to script on top of it.

Theoretically that's already possible, but the API is neither stable nor documented. See the dashboard source. I don't intend to make esphomeyaml an API because honestly the use-case seems quite seldom and I don't want to have to maintain that API.

OttoWinter, 2018-11-27:
Closing as most of this is possible using YAML anchors now. I also do not intend to use jinja as it's not made for YAML and some valid YAML could end up being interpreted by jinja2 into something else.

closed 2018-11-27 by OttoWinter