esphome/esphome #68: LICENSE

sh0rez, 2018-06-26:
Esphomeyaml is missing a LICENSE file or any other kind of declaration.
Is this intended? Should all rights be reserved? And does this comply with the Apache-License of homeassistant?

frog32, 2018-07-12:
Not having a license means I am legally not able to extend this and not having a CLA would prevent me from being able to file a pull request. Some of us have employers that require their employees to be a little picky here so if this is intended to be open please add those files.

OttoWinter, 2018-07-28:
Yes I understand. esphomeyaml is published under the MIT license as it says in the setup.py file and on pypi.org (https://pypi.org/project/esphomeyaml/#description). I will add a top-level LICENSE file asap.

------

esphomelib, the C++ base for esphomeyaml however is published under GPLv3 as seen here: https://github.com/OttoWinter/esphomelib/blob/master/LICENSE - the reason for this is that I don't really care about anybody copying the python part, as it's honestly quite ugly code rn. The C++ base however was a lot of work, especially all the sensor integrations.

frog32, 2018-07-29:
Unfortunately that would cause everyone who uses esphomelib to be in violation with the GPLv3 since some underlying libraries are not available in source code and espressif is not able to provide them. Please correct me if I am wrong but as I understand it using esphomelib is not possible if it is under GPLv3.

See https://github.com/espressif/ESP8266_RTOS_SDK/issues/1

frog32, 2018-07-29:
Maybe that is not a concern but at least esphomeyaml should inform it's users about the fact that their code (configuration) will be linked or combined with GPLv3 code and its implications.
At the same time you might want to provide an alternative option by relicensing it under something more permissive for people that are willing buy such a licence from you.

OttoWinter, 2018-07-31:
@frog32 Oh noes :( that's not good. I probably should have put more thought into picking the correct license (and not just looking at [this page](https://choosealicense.com/licenses/gpl-3.0/) 😀 )

That being said, I have no problem moving esphome*lib* under MIT too. I mean the code isn't really worth copying anyway 😂. But before I do that, I guess re-licensing would require me to ask any contributors about the change, right? From what I've seen you know way more about this than I do :)