esphome/esphome #70: Remember output state on reboot

TheCellMC, 2018-06-28:
Here's the problem:
I'm running esphomeyaml on an ESP8266 to control some of my lights. The ESP8266 is connected to 4 wall switches and 3 relays. I've used automations to control my relays with my switches. However, when the ESP8266 reboots because of a power issue, it closes all of the relays (which are set up using the GPIO Switch component). This results in all of the lights turning on. I don't know if this is caused by all of the GPIO Switch pins being pulled high or by the GPIO Binary Sensor triggering the automations on reboot. Would it be possible to make the ESP8266 remember the state of the outputs and inputs so that it can restore their state on a reboot ?

Landrash, 2018-06-30:
Since teh dev is on vacation I'll drop the transcript from the chat about a similar question.

```

jlaw - 06/18/2018
@OttoWinter my template sensor isn’t retaining its state between upgrades. is that expected?
config is identical to your suggestion above which got around the circular dependency

OttoWinter - 06/18/2018
On an ESP8266 yes. On an ESP32 no

jlaw - 06/18/2018
why’s that? does it not use the mqtt retain?

OttoWinter - 06/18/2018
Mqtt retain is only for states, not commands. The ESP32 has a special non voltage storage partition for these data stores. The esp8266 only has the "spiffs" which takes roughly 20ms per update, too much for esphomelib.

It's not using MQTT retain for commands because that has side effects. For example if the switch is turned on via HA and then turned off internally, the switch would still receive the "ON" retained message from HA on reboot even though it should restore "OFF".
+ state restoration should not depend on a connection like MQTT would
```

OttoWinter, 2018-07-28:
So, as per above comment I'm not a huge fan of storing the persistent state on the flash memory of the ESP. Especially on the ESP8266 getting state preservation to work is pretty hard without having some pretty big time side effects.

My idea for fixing this is on a more fundamental level: Reducing the amount of reboots. No reboots -> no state preservation problems ;) Most reboots in esphomelib are caused by pesky TCP stack errors that I still haven't fully figured out. In your case ("power issue") working more on making esphomelib's WiFi more stable will of course not solve your problem.

What *would* work is just setting up the switches as *inverted*. That way, when the node reboots the lights will be off, not on. For example like this:

```yaml
switch:
  - platform: gpio
    pin:
      number: 13
      inverted: True
    name: ...
```

Alternatively, you could just use the `on_boot` trigger to do any initialization on boot: https://esphomelib.com/esphomeyaml/components/esphomeyaml.html#esphomeyaml-on-boot

engrbm87, 2018-08-17:
Is it possible to change the on boot behavior so that the device will check the state topic for the last known state and switch the output state accordingly?

hknz, 2018-10-30:
I'm in a similar situation, I'm running esphomelib on an ESP8266.
I have just started out with home automation and I'm still in the early development stages.

I've started with the garage door example (https://esphomelib.com/esphomeyaml/cookbook/garage-door.html) with a reed switch too, which all works great. 
However whenever there's a power cycle the relay goes high. As there's an action associated with the relay, the garage door will open/close.
I've already tried invert and the same result happens on boot (Will try the on_boot trigger priority next).

When the ESP boots is it possible to make the cover default to...... payload_stop: "STATE"
As there isn't any action associated with this, the relay/door won't open/close?

OttoWinter, 2018-11-02:
@hknz In esphomelib 1.9.0 there will be a `power_on_value` option to have exactly that behavior.

escoand, 2018-11-02:
Am I right that this defines a value which is always set when power on? If I understand the request right this is not what was intended.

Currently my light is switching multiple times OFF every evening because the wifi connection dropped. Switching the light always ON after reboot is not better in this case. It's just another bad variant. Restoring the previous state so I don't even notice the reboot would be the best solution.

I understand the problem with the bad wifi stack but a dropped connection can always happen no matter how stable the stack is.

How does the reboot loop check work? Isn't this a similar problem? The device was running seconds ago...

escoand, 2018-11-02:
Shouldn't something like this work? But it's not...

```yaml
mqtt:
  broker: 192.168.1.45
  on_message:
    topic: licht_sofa/light/sofa/state
    then:
      - lambda: !lambda >-
          if(parse_on_off(x.c_str()))
            id(light).turn_on();
```

escoand, 2018-11-04:
Got a working solution:

```yaml
mqtt:
  broker: 192.168.1.45
  on_message:
    topic: licht_sofa/light/sofa/state
    then:
      - lambda: !lambda |-
          static int fresh_boot = 1;
          if(fresh_boot) {
              if(strcasecmp(x.c_str(), "{\"state\":\"on\"}") == 0) {
                  ESP_LOGI("boot", "restore last state: on");
                  id(light).turn_on().perform();
              } else {
                  ESP_LOGI("boot", "restore last state: off");
                  id(light).turn_off().perform();
              }
              fresh_boot = 0;
          }
```

brandond, 2018-11-05:
If the messages from HASS are set to retain, shouldn't esphomelib pick up the desired states when it connects and subscribes to the topics, and set them to the appropriate state automatically?

escoand, 2018-11-05:
For me this not seem to work. Or what's going on when light switches off occasionally without any reason?

brandond, 2018-11-05:
Are you using the built-in MQTT server, or something external? If you're using something external like Mosquitto, do you have persistence enabled and stored on a volume that doesn't get wiped when the container restarts?

escoand, 2018-11-05:
Yes, I use mosquitto and it is correctly persisting the messages.

Don't know exactly what happens. My current problem is that my router loses internet connection and apparently restart his whole wifi stack. This lasts some seconds, enough esphomelib device looses wifi connection and restarts. The light goes off and never lights up. But with my config above I can reset the previous state.

Is there a way to debug without using serial connection?

brandond, 2018-11-05:
you can tail logs via mqtt; the client should buffer enough to send you most of the logs from the boot sequence, even if the stuff before that is lost.

escoand, 2018-11-06:
Here is some logging recorded by my desktop pc.
Why is the light switching off at boot instead of waiting for the state topic?

```
# re-flash with verbose logging
licht_sofa/debug [I][ota:349]: OTA update finished!
licht_sofa/debug [I][mqtt.client:133]: MQTT Connected!
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/status' payload='online' retain=1)
licht_sofa/status online
licht_sofa/debug [C][light.mqtt_json:020]: Setting up MQTT light...
licht_sofa/debug [D][mqtt.client:188]: Subscribing to topic='licht_sofa/light/sofa/command' qos=0 with JSON...
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"OFF"}' retain=1)
licht_sofa/light/sofa/state {"state":"OFF"}
licht_sofa/debug [V][mqtt.component:067]: 'Sofa': Sending discovery...
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='homeassistant/light/licht_sofa/sofa/config' payload='{"flash":true,"name":"Sofa","platform":"mqtt_json","state_topic":"licht_sofa/light/sofa/state","command_topic":"licht_sofa/light/sofa/command","availability_topic":"licht_sofa/status","unique_id":"ESPlightsofa","device":{"identifiers":"2c3ae82f5a1d","name":"licht_sofa","sw_version":"esphomelib v1.10.0-dev Nov  6 2018, 20:45:33","model":"PLATFORMIO_ESP01_1M","manufacturer":"espressif"}}' retain=1)
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"OFF"}' retain=1)
licht_sofa/light/sofa/state {"state":"OFF"}
licht_sofa/debug [I][application:092]: You're running esphomelib v1.10.0-dev compiled on Nov  6 2018, 20:45:33
licht_sofa/debug [I][application:101]: Running through first loop()
licht_sofa/debug [I][application:118]: First loop finished successfully!
# this is my functionality from above, without this the light would stay off
licht_sofa/debug [I][restore_mqtt:046]: restore last state: on {"state":"ON"}
licht_sofa/debug [D][light.state:363]: 'Sofa' Turning ON
licht_sofa/debug [V][component:105]: Removing old time function send.
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"ON"}' retain=1)
licht_sofa/light/sofa/state {"state":"ON"}
licht_sofa/debug [I][ota:157]: Boot seems successful, resetting boot loop counter.
licht_sofa/status offline
licht_sofa/debug [I][mqtt.client:133]: MQTT Connected!
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/status' payload='online' retain=1)
licht_sofa/status online
licht_sofa/debug [C][light.mqtt_json:020]: Setting up MQTT light...
licht_sofa/debug [D][mqtt.client:188]: Subscribing to topic='licht_sofa/light/sofa/command' qos=0 with JSON...
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"OFF"}' retain=1)
licht_sofa/light/sofa/state {"state":"OFF"}
licht_sofa/debug [V][mqtt.component:067]: 'Sofa': Sending discovery...
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='homeassistant/light/licht_sofa/sofa/config' payload='{"flash":true,"name":"Sofa","platform":"mqtt_json","state_topic":"licht_sofa/light/sofa/state","command_topic":"licht_sofa/light/sofa/command","availability_topic":"licht_sofa/status","unique_id":"ESPlightsofa","device":{"identifiers":"2c3ae82f5a1d","name":"licht_sofa","sw_version":"esphomelib v1.10.0-dev Nov  6 2018, 20:45:33","model":"PLATFORMIO_ESP01_1M","manufacturer":"espressif"}}' retain=1)
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"OFF"}' retain=1)
licht_sofa/light/sofa/state {"state":"OFF"}
licht_sofa/debug [I][application:092]: You're running esphomelib v1.10.0-dev compiled on Nov  6 2018, 20:45:33
licht_sofa/debug [I][application:101]: Running through first loop()
licht_sofa/debug [I][application:118]: First loop finished successfully!
# this is my functionality from above, without this the light would stay off
licht_sofa/debug [I][restore_mqtt:046]: restore last state: on {"state":"ON"}
licht_sofa/debug [D][light.state:363]: 'Sofa' Turning ON
licht_sofa/debug [V][component:105]: Removing old time function send.
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"ON"}' retain=1)
licht_sofa/light/sofa/state {"state":"ON"}
licht_sofa/status offline
licht_sofa/debug [I][mqtt.client:133]: MQTT Connected!
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/status' payload='online' retain=1)
licht_sofa/status online
licht_sofa/debug [V][mqtt.component:067]: 'Sofa': Sending discovery...
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='homeassistant/light/licht_sofa/sofa/config' payload='{"flash":true,"name":"Sofa","platform":"mqtt_json","state_topic":"licht_sofa/light/sofa/state","command_topic":"licht_sofa/light/sofa/command","availability_topic":"licht_sofa/status","unique_id":"ESPlightsofa","device":{"identifiers":"2c3ae82f5a1d","name":"licht_sofa","sw_version":"esphomelib v1.10.0-dev Nov  6 2018, 20:45:33","model":"PLATFORMIO_ESP01_1M","manufacturer":"espressif"}}' retain=1)
licht_sofa/debug [V][mqtt.client:204]: Publish(topic='licht_sofa/light/sofa/state' payload='{"state":"ON"}' retain=1)
licht_sofa/light/sofa/state {"state":"ON"}
licht_sofa/debug [I][ota:157]: Boot seems successful, resetting boot loop counter.
```

escoand, 2018-11-07:
@OttoWinter is it possible to wait maybe 5s for the state topic and use this as initial state?

OttoWinter, 2018-11-08:
@escoand The problem with your "home assistant sends retained messages" approach is probably that when Home Assistant turns on the light, an ON message is published and retained. When the node then turns off the light by another mean, the command topic will still read the retained ON message.

However, this won't be a problem in the future: https://github.com/OttoWinter/esphomelib/pull/258 :P. I found a way to store data in RTC memory on the ESP8266 which does not break OTA.

escoand, 2018-11-08:
@OttoWinter this is really good news. I can't wait to test this. And I think my wife can't wait to not need complain the light switches. ;-)

escoand, 2018-11-14:
It's working and not switching off anymore. But I realized a little flickering maybe half a second or so. ATM I don't have logs...

OttoWinter, 2018-11-14:
@escoand Is that using the `light` component? For state restoration the default values for the transition length are used. So if you haven't defined a `default_transition_length` the lights will fade to on in 1 second on bootup

escoand, 2018-11-15:
@OttoWinter You nailed it, but I use a binary light which don't accepts this value. But I already realized the transition message in the logs.

Harriethijssen, 2018-12-18:
@OttoWinter, you previously stated: 

> `The esp8266 only has the "spiffs" which takes roughly 20ms per update, too much for esphomelib.`

Can you please explain why 20ms is too much? Using RTC is ok but if I understand this memory it will not persist a power failure. I live in an area where power interuptions are common and want to have the choice how a sensor should initialize itself after an interrupt (I.e. “off”, “on”, “last state”, and maybe others based on time of day for instance). 
I have several philips hue lights, and find it irritating that all lights turn on after a power interruption..

I’m not sure why 20ms matters too much after the state of a sensor has changed (unless the power interruption takes place during those 20ms).

Thanks for elaborating.

OttoWinter, 2018-12-19:
@Harriethijssen 20ms is too much because esphomelib needs to work in real time. For example if a light in the middle of a transition you would quickly notice it. But another issue I apparently forgot to mention there is that many flash writes can quickly wear out the flash. This means that there would have to be an entire library on top of the flash storage that automatically rotates when a block becomes bad.

I guess I can introduce an `power_on_state:` option to the GPIO switch for these use cases. 
 - `OFF` - always initialize as off on boot
 - `ON` - always initialize as on on boot.
 - `RESTORE_ON` - try to restore and use ON as backup.
 - `RESTORE_OFF` - try to restore and use OFF as backup (default)

marcinkowalczyk, 2019-02-13:
@OttoWinter any luck in impementing this ?

brandond, 2019-02-13:
@marcinkowalczyk see restore_mode: https://esphome.io/components/switch/gpio.html

GreatSnoopy, 2019-03-21:
I am dropping in on this quite late. Why would not the mqtt retain work? As i see it:
- Whenever esphome switches the state of the relay, it also publishes the new state on its dedicated topic, with retain (and it should not even matter who exactly requested the relay change, either the API or a physical button
- if a power failure occurs, esphome should first try to read it's topic so if there is data it should then set the state of the relay based on the data from the topic. (this phase only happens at boot, basically after a power failure)

Further, it is much easier to make sure that the retained messages persist, by either permanent storage or a single UPS on the mqtt server rather than making sure that all the devices are protected from power drops. You cannot guard all of them, but can guard your broker.
Obviously, if no message can be retrieved from the topic, there should also be a hardcoded default.

This approach means also that the local flash would not be trashed by lots of rewrites.
Finally, that also potentially allows one to change the preferred status even if the device is offline: one can simply override the retained message so that when the device powers up again it will pick the actual desired state.

OttoWinter, 2019-03-22:
@GreatSnoopy No, as written here retained messages do have big drawbacks too. Biggest one being one of the biggest features in esphome: what if an automation changes the state of a relay while it is not connected to MQTT - it can't reliably know which one to choose. See https://github.com/esphome/esphome/pull/227#discussion_r232460220 - restoring state from MQTT will not happen, I have looked into it already and it's not possible to pull off reliably.

I also don't think you will "trash" the flash that quickly under normal use. You would have to perform lots of write operations before that.

GreatSnoopy, 2019-03-25:
@OttoWinter if you are confident that lots of rewrites for the flash is not an issue, especially in the context of an automation that will trigger multiple times per day (think motion sensor) -> then this is a good enough method to persist state.
However, the mqtt retained messages were made especially for this scenario and while I do agree with you that it's quite undesirable to not be able to connect to mqtt when needed, I feel that it is much easier to achieve mqtt availability and it may not be an issue anyway. (I think the two methods along with some trivial versioning could be employed to rely on mqtt if available or the local state cache in flash memory otherwise)

But thanks for the clarification, anyway !