esphome/esphome #71: DHT error

mefizto, 2018-06-30:
Hi again
I get the following error every time with the dht.

```
INFO [esphomeyaml.config] Reading configuration...
INFO [esphomeyaml.mqtt] Starting log output from escritorio/debug
[21:09:04][I][mqtt.client:132]: MQTT Connected!
[21:09:04][C][sensor.dht:031]: Setting up DHT...
[21:09:04][C][esphal:053]:     GPIO Pin 16 with mode INPUT
[21:09:04][C][component:239]:     Update interval: 15000ms
[21:09:04][C][light.mqtt_json:022]: Setting up MQTT light...
[21:09:04][D][mqtt.client:186]: Subscribing to topic='escritorio/light/luz_de_escritorio/command' qos=0 with JSON...
[21:09:04][C][sensor.mqtt:025]: Setting up MQTT Sensor 'Humedad cerca del escritorio: '...
[21:09:04][C][sensor.mqtt:027]:     Expire After: 675s
[21:09:04][C][sensor.mqtt:029]:     Unit of Measurement: '%'
[21:09:04][C][sensor.mqtt:030]:     Accuracy Decimals: 0
[21:09:04][C][sensor.mqtt:031]:     Icon: 'mdi:water-percent'
[21:09:04][C][sensor.mqtt:025]: Setting up MQTT Sensor 'Temperatura cerca del Escritorio: '...
[21:09:04][C][sensor.mqtt:027]:     Expire After: 675s
[21:09:04][C][sensor.mqtt:029]:     Unit of Measurement: '°C'
[21:09:04][C][sensor.mqtt:030]:     Accuracy Decimals: 1
[21:09:04][C][sensor.mqtt:031]:     Icon: ''
[21:09:04][I][application:090]: Running through first loop()
[21:09:04][C][sensor.dht:045]: Auto-detected model DHT22. Override with the 'model:' configuration option
[21:09:04][I][application:106]: First loop finished successfully!
**[21:09:17][W][sensor.dht:104]: Rising edge for bit -1 failed!
[21:09:17][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0**
```

My yaml loos like:

```
output:
  - platform: esp8266_pwm
    pin: D0
    id: 'output_component1'
  - platform: esp8266_pwm
    pin: D3
    id: 'output_component2'
  - platform: esp8266_pwm
    pin: D4
    id: 'output_component3'
      
light:
  - platform: rgb
    name: "Luz de Escritorio"
    red: output_component1
    green: output_component2
    blue: output_component3

sensor:
  - platform: dht
    pin:
        number: D0
        mode: INPUT
    model: AUTO_DETECT
    temperature: 
        name: "Temperatura cerca del Escritorio: "
    humidity:
        name: "Humedad cerca del escritorio: "
    update_interval: 15s
```
I tested the sensor with an "arduino nano" and it works correctly.

mefizto, 2018-06-30:
I had a typo on pin of the sensor try to use the pin D0 that I had already used but I use D2 however I have the same output in the log ,. I used AUTO_DETECT and detects a DHT22 this is correct and now I try specifying the DHT22 model and I get the same result on the log

mefizto, 2018-07-16:
If anyone can help me please, i tried another DHT sensor a DHT11 and a DHT22 and i get the same

 ```
error i share the log and yaml from my module:
INFO [esphomeyaml.espota] Waiting for result...
INFO [esphomeyaml.espota] Result: OK
INFO [esphomeyaml.__main__] Successfully uploaded program.
INFO [esphomeyaml.mqtt] Starting log output from pruebas/debug
[19:29:51][I][mqtt.client:132]: MQTT Connected!
[19:29:51][C][sensor.dht:031]: Setting up DHT...
[19:29:51][C][esphal:053]:     GPIO Pin 13 with mode INPUT
[19:29:51][C][component:239]:     Update interval: 15000ms
[19:29:51][C][light.mqtt_json:022]: Setting up MQTT light...
[19:29:51][D][mqtt.client:186]: Subscribing to topic='pruebas/light/leds_escritorio/command' qos=0 with JSON...
[19:29:51][C][light.mqtt_json:022]: Setting up MQTT light...
[19:29:51][D][mqtt.client:186]: Subscribing to topic='pruebas/light/keyboard_lights/command' qos=0 with JSON...
[19:29:51][C][sensor.mqtt:025]: Setting up MQTT Sensor 'Living Room Humidity'...
[19:29:51][C][sensor.mqtt:027]:     Expire After: 675s
[19:29:51][C][sensor.mqtt:029]:     Unit of Measurement: '%'
[19:29:51][C][sensor.mqtt:030]:     Accuracy Decimals: 0
[19:29:51][C][sensor.mqtt:031]:     Icon: 'mdi:water-percent'
[19:29:51][C][sensor.mqtt:025]: Setting up MQTT Sensor 'Living Room Temperature'...
[19:29:51][C][sensor.mqtt:027]:     Expire After: 675s
[19:29:51][C][sensor.mqtt:029]:     Unit of Measurement: '°C'
[19:29:51][C][sensor.mqtt:030]:     Accuracy Decimals: 1
[19:29:51][C][sensor.mqtt:031]:     Icon: ''
[19:29:51][C][binary_sensor.mqtt:024]: Setting up MQTT binary sensor 'Toggle Button 2'...
[19:29:51][C][binary_sensor.mqtt:024]: Setting up MQTT binary sensor 'Toggle Button'...
[19:29:51][C][binary_sensor.mqtt:024]: Setting up MQTT binary sensor 'PIR Sensor'...
[19:29:51][C][binary_sensor.mqtt:026]:     Device Class: 'motion'
[19:29:51][I][application:090]: Running through first loop()
[19:29:51][W][sensor.dht:118]: Falling edge for bit -1 failed!
[19:29:51][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[19:29:51][I][application:106]: First loop finished successfully!
[19:29:59][W][sensor.dht:118]: Falling edge for bit -1 failed!
[19:29:59][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[19:30:14][W][sensor.dht:118]: Falling edge for bit -1 failed!
[19:30:14][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[19:30:29][W][sensor.dht:118]: Falling edge for bit -1 failed!
[19:30:29][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[19:30:44][W][sensor.dht:118]: Falling edge for bit -1 failed!
[19:30:44][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[19:30:59][W][sensor.dht:118]: Falling edge for
```

```
esphomeyaml:
  name: pruebas
  platform: ESP8266
  board: nodemcuv2 

wifi:
  ssid: !secret wifi
  password: !secret psw
  
  manual_ip:
    static_ip: 192.66.6.104
    gateway: 192.66.6.1
    subnet: 255.255.255.0

mqtt:
  broker: '192.66.6.110'
  username: !secret usuario
  password: !secret contraseña

# Enable logging
logger:

ota:
  safe_mode: True
  password: !secret contraseña

output:
  - platform: esp8266_pwm
    pin: D2
    id: 'color_red'
  - platform: esp8266_pwm
    pin: D4
    id: 'color_green'
  - platform: esp8266_pwm
    pin: D5
    id: 'color_blue'
  - platform: esp8266_pwm
    pin: D3
    id: "atx0"

light:
  - platform: rgb
    name: "LEDs escritorio"
    red: color_red
    green: color_green
    blue: color_blue
    id: led1
  - platform: monochromatic
    name: "Keyboard Lights"
    output: atx0
    id: led0
    
binary_sensor:
  - platform: gpio
    pin: D6
    name: "PIR Sensor"
    device_class: motion
  - platform: gpio
    pin: D1
    name: "Toggle Button"
    on_press:
      then:
        - light.toggle:
            id: led1
  - platform: gpio
    pin: D3
    name: "Toggle Button 2"
#    on_press:
 #     then:
  #      - light.toggle:
   #         id: led0
            
power_supply:
  - id: 'atx_power_supply'
    pin:
      number: D8
      inverted: true
      
sensor:
  - platform: dht
    pin: 
        number: D7
        mode: INPUT
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    model: DHT11
    update_interval: `15s
```

mefizto, 2018-07-16:
i did somtheing and now my logs show:

```INFO [esphomeyaml.config] Reading configuration...
INFO [esphomeyaml.mqtt] Starting log output from pruebas/debug
[22:31:05][E][sensor.dht:151]: Checksum invalid: 4!=234
[22:31:05][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0
[22:31:20][E][sensor.dht:151]: Checksum invalid: 4!=229
[22:31:20][W][sensor.dht:059]: Invalid readings! Please check your wiring (pull-up resistor, pin_ number) and consider manually specifying the DHT model using the model option. Error code: 0```

I use a resistor pullup on data pin, I dont know what else to do, please help

carlosmgr, 2018-07-23:
That's a problem that i'm also having, but i think it's something related to the sensor or the board. 
On one of my boards i manualy change from  AUTO_DETECT to DHT11 and the problem was solved, on the other one it did not.
I'm gonna change the sensor to see the output, but in my case i thing is samething related to board since everytime i turn on the relay this errors are constant.

carlosmgr, 2018-07-23:
I just spend the last hour testing, i changed every thing except the board, and when i turn on the light the DHT11 gives me the checksum error. The  intriguing part is that if i disconnect the light cable and leave the relay on, i get no errors....  

@mefizto  what board are you using and why are you using input?
Are you using the dht sensor with a transistor between the vcc and data? 
 
The  "Falling edge for bit -1 failed!" errors it's when do not have anything connected.

zmrow, 2018-07-24:
This feels like a hardware issue rather than a software issue.  Perhaps this is not the correct forum to continue this discussion.  However that is up to the package maintainer.  :)

If the maintainer feels that continuing the discussion here is productive, more details on the hardware setup are necessary.  Board model, DHT sensor model, circuit diagram/picture, etc.

ezbe, 2018-07-24:
works fine for me with that code and AM2302
```sensor:
  - platform: dht
    pin: GPIO1
    temperature:
      name: "Living room temperature"
    humidity:
      name: "Living room humidity"
    model: AM2302
    update_interval: 15s```

ezbe, 2018-07-24:
possibly your problem is here ```    pin: 
        number: D7
        mode: INPUT```

OttoWinter, 2018-07-28:
So I don't really know how to help with your problem @mefizto unfortunately, from what I see it kind of looks like a hardware issue to me. Maybe you can share your wiring?

A bit of clarification:

* The way `AUTO_DETECT` works is that on the first measurement, esphomelib basically just tries to get the data as if it was a DHT22. If it doesn't work in DHT22 mode, esphomelib just assumes it would be an DHT11.
* `Falling edge for bit -1 failed!` means that the DHT doesn't respond to the measurement request. To tell the DHT to start a measurement, esphomelib first pulls down the data line to ground for a certain time and then waits for the sensor to start sending data. If the sensor never responds, that error appears.
* 
```yaml
pin: 
   number: D7
   mode: INPUT
```

The pin shouldn't be declared as an input, just a simple `pin: D7` should work. Declaring it as an input should however just mess up the very first measurement if I'm correct.

mastershake8241, 2018-08-04:
I made a post about this and the DHT22. I had to hook my ground from the sensor to another pin like D0 and add that pin as a switch. My DHT22's work everytime now.

OttoWinter, 2018-11-27:
Closing this issue as I think the DHT integration has gotten better now. At least I've not seen these errors pop up in a while on github.

closed 2018-11-27 by OttoWinter