esphome/esphome #72: OTA: No result - but succeeds

simonszu, 2018-06-30:
Currently i am experimenting with OTA feature. It seems to work, but i have a question. This is the output of `esphomeyaml run`:
```
INFO [esphomeyaml.espota] Starting on 0.0.0.0:20574
INFO [esphomeyaml.espota] Upload size: 349248
INFO [esphomeyaml.espota] Sending invitation to schlafzimmerklima.home.simonszu.de 

INFO [esphomeyaml.espota] Authenticating...
INFO [esphomeyaml.espota] OK
INFO [esphomeyaml.espota] Waiting for device...
Uploading: [============================================================] 100% Done...

INFO [esphomeyaml.espota] Waiting for result...
ERROR [esphomeyaml.espota] No Result!
```
However, upload seems to succeed, the device is back online and sends stuff to the MQTT broker. What result does esphomeyaml expect, and what could i do to make the device send this result to esphomeyaml?

Landrash, 2018-06-30:
Depending on your log level you will have a log returned in the console.

OttoWinter, 2018-07-28:
That's an error of the OTA library esphomelib uses. AFAIK this mainly happens when the WiFi connection is not very strong. Then the OTA result doesn't reach esphomeyaml in time before the node reboots. As the OTA still does technically work and it's an error of the Arduino framework I don't intend to fix it myself, maybe the Arduino guys will fix it soon.

OttoWinter, 2018-11-27:
Since 1.9 has its own OTA update engine which uses the proper TCP/socket flags (like TCP_NODELAY) which should resolve this issue.

closed 2018-11-27 by OttoWinter