esphome/esphome #74: Panic with binary_sensor template

nunofgs, 2018-07-02:
Hi all, I'm getting a panic with a binary_sensor template on version 1.7.0. I managed to reproduce it with just:

```yaml
binary_sensor:
  - platform: template
    name: "Left Parking Spot"
    lambda: >-
      return true;
```

Here's the log output:

```
[22:57:14][C][binary_sensor.mqtt:024]: Setting up MQTT binary sensor 'Left Parking Spot'...
[22:57:14][D][debug:034]: esphomelib version 1.7.0
[22:57:14][D][debug:036]: Free Heap Size: 28752 bytes
[22:57:14][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[22:57:14][D][debug:131]: Chip ID: 0x00874CB1
[22:57:14][D][debug:132]: SDK Version: 2.2.1(cfd48f3)
[22:57:14][D][debug:133]: Core Version: 2_4_1
[22:57:14][D][debug:134]: Boot Version=31 Mode=1
[22:57:14][D][debug:135]: CPU Frequency: 80
[22:57:14][D][debug:136]: Flash Chip ID=0x001640E0
[22:57:14][D][debug:137]: Reset Reason: External System
[22:57:14][D][debug:138]: Reset Info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[22:57:14][I][application:090]: Running through first loop()
[22:57:14]
[22:57:14]Exception (2):
[22:57:14]epc1=0x3fff3cd4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff3cd4 depc=0x00000000
[22:57:14]
[22:57:14]ctx: cont 
[22:57:14]sp: 3fff1d20 end: 3fff1f40 offset: 01a0
[22:57:14]
[22:57:14]>>>stack>>>
[22:57:14]3fff1ec0:  3fffdad0 3fff3644 3fff3754 3fff3cd4  
[22:57:14]3fff1ed0:  3fff3c90 00000001 3fff3cd4 40204125  
[22:57:14]3fff1ee0:  40106fc8 00000000 00000002 4020246c  
[22:57:14]3fff1ef0:  3fff3ca0 00000000 40106fc8 00000000  
[22:57:14]3fff1f00:  40106fc8 00000000 3fff3cd4 3fff0f10  
[22:57:14]3fff1f10:  3fffdad0 00000000 3fff0f08 4020221c  
[22:57:14]3fff1f20:  feefeffe 00000000 3fff0f08 40209574  
[22:57:14]3fff1f30:  feefeffe feefeffe 3fff0f20 40100700  
[22:57:14]<<<stack<<<
[22:57:14]
[22:57:14] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[22:57:14]
[22:57:14]load 0x4010f000, len 1384, room 16 
[22:57:14]tail 8
[22:57:14]chksum 0x2d
[22:57:14]csum 0x2d
[22:57:14]v614f7c32
[22:57:14]~ld
[22:57:15][I][log_component:056]: Log initialized
[22:57:15][C][ota:182]: There have been 1 suspected unsuccessful boot attempts.
```

Any thoughts?

nunofgs, 2018-07-24:
@OttoWinter is this project still being worked on?

brandond, 2018-07-24:
@nunofgs I believe he's on holiday.

nunofgs, 2018-07-24:
@brandond Ah, wasn't aware. All's good then :)

I ended up taking some time to debug this and found the cause.

closed 2018-07-28 by OttoWinter