esphome/esphome #75: Error when outputting logs after first boot

nunofgs, 2018-07-02:
Hi all. When I first upload my configuration to a nodemcu device through USB, it manages to attach and display the log output correctly. If I then change the configuration and re-upload, the logs never show up.

The output in my console shows:

```
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/tornado/ioloop.py", line 759, in _run_callback
    ret = callback()
  File "/usr/local/lib/python2.7/site-packages/tornado/stack_context.py", line 276, in null_wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/tornado/ioloop.py", line 780, in _discard_future_result
    future.result()
  File "/usr/local/lib/python2.7/site-packages/tornado/concurrent.py", line 260, in result
    raise_exc_info(self._exc_info)
  File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 1113, in run
    yielded = self.gen.send(value)
  File "/usr/local/lib/python2.7/site-packages/esphomeyaml/dashboard/dashboard.py", line 71, in redirect_stream

  File "/usr/local/lib/python2.7/site-packages/tornado/websocket.py", line 255, in write_message
    message = tornado.escape.json_encode(message)
  File "/usr/local/lib/python2.7/site-packages/tornado/escape.py", line 81, in json_encode
    return json.dumps(value).replace("</", "<\\/")
  File "/usr/local/Cellar/python@2/2.7.15/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 244, in dumps
    return _default_encoder.encode(obj)
  File "/usr/local/Cellar/python@2/2.7.15/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py", line 207, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/Cellar/python@2/2.7.15/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/encoder.py", line 270, in iterencode
    return _iterencode(o, 0)
UnicodeDecodeError: 'utf8' codec can't decode byte 0x84 in position 18: invalid start byte
```

Now, the interesting part is that if I physically disconnect and reconnect the USB cable, the next upload works. I can see the logs again. It looks like it's just subsequent uploads.

OttoWinter, 2018-07-28:
Oh man, these `UnicodeDecodeError` are driving me nuts. I can't wait to be able to use python 3 for this project :) The issue here is (probably) that while the node boots up, the UART pins are floating for a split second and the serial monitor is receiving some non-UTF8 data, therefore causing the logs to fail.

I don't know if it's the case here, but if you are trying to view the USB/UART logs *while* uploading via USB it will fail, as only one device at a time can access the USB port without messing things up.

OttoWinter, 2018-09-27:
Should be fixed in 1.8.0

closed 2018-09-27 by OttoWinter