esphome/esphome #76: [Feature Request] HX711 Loadcell amplifier compatibility

geelenbert, 2018-07-03:
I really would like support for the HX711 load cell amplifier.
The main goal would be to create a giant scale of my bed to use as occupancy sensor. 
I would like to set some sort of threshold (difference between last reported value) to keep the updates to Home assistant a little bit limited. Also an amplification/calculation factor for calculating Kg.

nphil, 2018-07-29:
seconded!

OttoWinter, 2018-07-31:
Uhm sure, from what I see it's not too complicated to implement so I will try to maybe implement it soon

geelenbert, 2018-07-31:
That would be great!
I have loadcells and HX711 pcb’s available. 

Can test with ESP32 and ESP8622 modules.

pyrosmiley, 2018-08-07:
Definitely gonna bump this also! I have several projects looking to work with these and not having to write firmware from scratch for them would be wonderful

vincegre, 2018-08-15:
Any chance to get support for it soon ? ;)

cgtobi, 2018-08-16:
Looking at the past commits I think this has just been added.

https://github.com/OttoWinter/esphomeyaml/commit/5170a7cdf404f6a16babd73d61865b28a2520aa7#diff-fd5cebdd8634e56b5e3ff59e810a118b

OttoWinter, 2018-09-19:
Yes, it has been added and will be in the next release (for which I'm currently preparing). I have **not** tested this component as I don't have the hardware, but I'm pretty hopeful it will work as the protocol is dead simple.

Please open another issue if the HX711 integration has issues.

closed 2018-09-19 by OttoWinter