esphome/esphome #78: Getting Exception Error for NodeMCU ESP32-S When Running BLE Beacon

techwithjake, 2018-07-06:
After uploading and running BLE Beacon on two separate NodeMCU ESP32-S boards. I'm getting:

<pre>
[22:26:41]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[22:34:10]Core 0 register dump:
[22:34:10]PC      : 0x400d2fb8  PS      : 0x00060b30  A0      : 0x00000000  A1      : 0x3ffda580
[22:34:10]A2      : 0x21d1d0e2  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
[22:34:10]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x21d1d0e2  A9      : 0x3ffda530
[22:34:10]A10     : 0x00000048  A11     : 0x3f401ae4  A12     : 0x3f401c4c  A13     : 0x3ffda530
[22:34:11]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
[22:34:11]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[22:34:11]`
[22:34:11]Backtrace: 0x400d2fb8:0x3ffda580 0x7ffffffd:0x3ffda5e0
</pre>

A sample of my yaml file:

<pre>
esphomeyaml:
  name: living_room_ble
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: 'mySSID'
  password: 'myPass'

mqtt:
  broker: 'brokerIP'
  username: 'mqttUser'
  password: 'mqttPass'

# Enable logging
logger:

#ota:
#  safe_mode: True
#  password: 'otaPass'

esp32_ble_beacon:
  type: iBeacon
  uuid: 'beaconUUID'

status_led:
  pin: GPIO2
</pre>

These are two different NodeMCU ESP32-S boards but months apart. They are from "HiLetGo" on Amazon.

System is Windows 10 with latest updates on system and esphomeyaml. Tried uploading on different ports on the computer, both USB 2 and USB 3.

techwithjake, 2018-07-06:
I've changed to a different computer and WiFi network with a technically "different" MQTT broker (external URL vs internal IP) with the same results. Changed UUID as well on the off chance.

Only similar thing is name of the node "living_room_ble" and MQTT User and Pass.

yuanl, 2018-07-16:
+1 same here.

OttoWinter, 2018-07-28:
Hmm, illegal instruction usually means that the beacon thread exited prematurely because of some error. It *should* however print out that error before. Do you see an error or something else show up before that stacktrace?

yuanl, 2018-07-29:
Here's what I got.
```
[16:02:44][I][mqtt.client:095]: Connecting to MQTT...
[16:02:44][I][mqtt.client:132]: MQTT Connected!
[16:02:44][E][esp32_ble_beacon:078]: esp_bt_controller_init failed: 259
[16:02:44]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[16:02:44]Core 0 register dump:
[16:02:44]PC      : 0x400d36e4  PS      : 0x00060030  A0      : 0x00000000  A1      : 0x3ffdb960
[16:02:44]A2      : 0xce5d4390  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
[16:02:44]A6      : 0x00000000  A7      : 0x00000000  A8      : 0xce5d4390  A9      : 0x3ffdb910
[16:02:44]A10     : 0x00000048  A11     : 0x3f401bd5  A12     : 0x3f401cbe  A13     : 0x3ffdb910
[16:02:44]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
[16:02:44]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[16:02:44]
[16:02:44]Backtrace: 0x400d36e4:0x3ffdb960 0x7ffffffd:0x3ffdb9c0
[16:02:44]
[16:02:44]Rebooting...
```

GreenTentacle, 2018-07-30:
I'm getting the same situation; however this only started on builds after running a `platformio update`, so perhaps there's something that was deprecated or changed there ?

```
INFO [esphomeyaml.__main__] Starting log output from /dev/ttyUSB0 with baud rate 115200
[02:40:51]: Setting up WiFi...
[02:40:51][I][wifi:227]: WiFi Connecting to 'redacted'...
[02:40:51][C][wifi:244]:     STA Hostname: 'blebeacon'
[02:40:52][I][wifi:278]: WiFi connected!
[02:40:52][C][wifi:280]:     SSID: redacted
[02:40:52][C][wifi:282]:     BSSID: redacted
[02:40:52][C][wifi:283]:     Channel: 1
[02:40:52][C][wifi:284]:     IP Address: 192.168.73.211
[02:40:52][C][wifi:285]:     Subnet: 255.255.254.0
[02:40:52][C][wifi:286]:     Gateway: 192.168.73.1
[02:40:52][C][wifi:287]:     DNS1: 192.168.73.1
[02:40:52][C][wifi:288]:     DNS2: 0.0.0.0
[02:40:52][C][mqtt.client:024]: Setting up MQTT...
[02:40:52][C][mqtt.client:025]:     Server Address: redacted:1883
[02:40:52][C][mqtt.client:026]:     Username: ''
[02:40:52][C][mqtt.client:027]:     Password: ''
[02:40:52][C][mqtt.client:030]:     Client ID: 'homebeacon-redacted'
[02:40:52][C][mqtt.client:032]:     Discovery prefix: 'homeassistant'
[02:40:52][C][mqtt.client:033]:     Discovery retain: true
[02:40:52][I][mqtt.client:095]: Connecting to MQTT...
[02:40:52][I][mqtt.client:132]: MQTT Connected!
[02:40:52][C][esp32_ble_beacon:049]: Setting up ESP32 BLE beacon...
[02:40:52][I][application:090]: Running throuontroller_init fail[E][esp32_ble_beacon:078]: esp_bt_controller_init failed: 259
[02:40:52]
[02:40:52][I]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[02:40:52]Core 0 register dump:
[02:40:52]PC      : 0x400d2ffc  PS      : 0x00060930  A0      : 0x00000000  A1      : 0x3ffda6d0  
[02:40:52]A2      : 0x5a1a5ebd  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
[02:40:52]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x5a1a5ebd  A9      : 0x3ffda680  
[02:40:52]A10     : 0x00000048  A11     : 0x3f401ad4  A12     : 0x3f401c3c  A13     : 0x3ffda680  
[02:40:52]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000  
[02:40:52]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[02:40:52]
[02:40:52]Backtrace: 0x400d2ffc:0x3ffda6d0 0x7ffffffd:0x3ffda730
[02:40:52]
[02:40:52]Rebooting...
```

techwithjake, 2018-08-07:
This is what I get after doing a brand new installation and made sure everything was up to date.

```
[10:33:16]Rebooting...
[10:33:16]ets Jun  8 2016 00:22:57
[10:33:16]
[10:33:16]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:33:16]configsip: 0, SPIWP:0xee
[10:33:16]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:33:16]mode:DIO, clock div:2
[10:33:16]load:0x3fff0018,len:4
[10:33:16]load:0x3fff001c,len:952
[10:33:16]load:0x40078000,len:6084
[10:33:16]load:0x40080000,len:7944
[10:33:16]entry 0x40080310
[10:33:17][I][log_component:056]: Log initialized
[10:33:17][C][ota:182]: There have been 7 suspected unsuccessful boot attempts.
[10:33:17][I][application:048]: Running through setup()...
[10:33:17][C][wifi:031]: Setting up WiFi...
[10:33:17][I][wifi:227]: WiFi Connecting to '[REDACATED]'...
[10:33:17][C][wifi:244]:     STA Hostname: 'living_room_bt'
[10:33:18][I][wifi:278]: WiFi connected!
[10:33:18][C][wifi:280]:     SSID: [REDACATED]
[10:33:18][C][wifi:282]:     BSSID: [REDACATED]
[10:33:18][C][wifi:283]:     Channel: 1
[10:33:18][C][wifi:284]:     IP Address: 192.168.XXX.XXX
[10:33:18][C][wifi:285]:     Subnet: 255.255.255.0
[10:33:18][C][wifi:286]:     Gateway: 192.168.XXX.XXX
[10:33:18][C][wifi:287]:     DNS1: 205.171.3.26
[10:33:18][C][wifi:288]:     DNS2: 68.1.16.107
[10:33:18][C][ota:026]: Setting up OTA...
[10:33:18][C][ota:027]:     port: 3232
[10:33:18][C][ota:032]:     hostname: 'living_room_bt'
[10:33:18][W][ota:120]: Last Boot was an unhandled reset, will proceed to safe mode in 3 restarts
[10:33:18][C][mqtt.client:024]: Setting up MQTT...
[10:33:18][C][mqtt.client:025]:     Server Address: [REDACATED]
[10:33:18][C][mqtt.client:026]:     Username: '[REDACATED]'
[10:33:18][C][mqtt.client:027]:     Password: '[REDACATED]'
[10:33:18][C][mqtt.client:030]:     Client ID: 'living_room_bt-[REDACATED]'
[10:33:18][C][mqtt.client:032]:     Discovery prefix: 'homeassistant'
[10:33:18][C][mqtt.client:033]:     Discovery retain: true
[10:33:18][I][mqtt.client:095]: Connecting to MQTT...
[10:33:18][I][mqtt.client:132]: MQTT Connected!
[10:33:18][C][esp32_ble_beacon:049]: Setting up ESP32 BLE beacon...
[10:33:18][I][application:090]: Running througntroller_init fail
[10:33:18][I][application:106]: First loop finished successfully! 9[I][application:106]: First loop finished successfully!
[10:33:18]
[10:33:18]Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[10:33:18]Core 0 register dump:
[10:33:18]PC      : 0x400d3130  PS      : 0x00060a30  A0      : 0x00000000  A1      : 0x3ffdb7d0
[10:33:18]A2      : 0x3f824475  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
[10:33:18]A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3f824475  A9      : 0x3ffdb780
[10:33:18]A10     : 0x00000048  A11     : 0x3f401b44  A12     : 0x3f401cac  A13     : 0x3ffdb780
[10:33:18]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
[10:33:18]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[10:33:18]
[10:33:18]Backtrace: 0x400d3130:0x3ffdb7d0 0x7ffffffd:0x3ffdb830
[10:33:18]
```

OttoWinter, 2018-08-12:
Ah ok that's the culprit:

```bash
[E][esp32_ble_beacon:078]: esp_bt_controller_init failed: 259
```

That error message indicates that the BT module on the ESP32 is in an invalid state. The code used for the ESP32 BLE beacon/tracker code is pretty much the example code from the esp-idf code from the manufacturer, so it's probably not a direct problem of esphomelib. 

Could you try two things for me?

1. Enable the [debug component](https://esphomelib.com/esphomeyaml/components/debug.html) and see if the `BLE` and `BT` are in the features section. From what I know espressif also sells ESP32 chips without the BT module on board.
2. Try [using the latest arduino framework version](https://esphomelib.com/esphomeyaml/components/esphomeyaml.html#using-the-latest-arduino-framework-version). The latest arduino framework also uses the latest esp-idf SDK by espressif. Maybe espressif has fixed something recently...

yuanl, 2018-08-12:
1. I had enable debug then get this in feature.
```
Serial port /dev/cu.SLAB_USBtoUART
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
```
2. Use the latest framework
```
[23:11:44][I][mqtt.client:095]: Connecting to MQTT...
[23:11:44][I][mqtt.client:132]: MQTT Connected!
[23:11:44][C][component:239]:     Update interval: 15000ms
[23:11:44][C][sensor.bme280:095]: Setting up BME280...
[23:11:44][C][sensor.bme280:096]:     Address: 0x77
[23:11:45][C][sensor.bme280:126]:     Temperature Oversampling: 16x
[23:11:45][C][sensor.bme280:127]:     Pressure Oversampling: 16x
[23:11:45][C][sensor.bme280:128]:     Humidity Oversampling: 16x
[23:11:45][C][sensor.bme280:129]:     IIR Filter: OFF
[23:11:45][C][component:239]:     Update interval: 15000ms
[23:11:45][C][esp32_ble_beacon:049]: Setting up ESP32 BLE beacon...
[23:11:45][C][sensor.mqtt:025]: Setting up MQTntroller_init failed: 259..[E][esp32_ble_beacon:078]: esp_bt_controller_init failed: 259
[23:11:45]
[23:11:45][C][sensor.mqtt:027]:     Expire Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[23:11:45]Core 0 register dump:
[23:11:45]PC      : 0x400d3c90  PS      : 0x00060a30  A0      : 0x00000000  A1      : 0x3ffdbb30
[23:11:45]A2      : 0xe2215b7e  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
[23:11:45]A6      : 0x00000000  A7      : 0x00000000  A8      : 0xe2215b7e  A9      : 0x3ffdbae0
[23:11:45]A10     : 0x00000048  A11     : 0x3f402088  A12     : 0x3f4021f0  A13     : 0x3ffdbae0
[23:11:45]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
[23:11:45]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[23:11:45]
[23:11:45]Backtrace: 0x400d3c90:0x3ffdbb30 0x7ffffffd:0x3ffdbb90
[23:11:45]
[23:11:45]Rebooting...
```

techwithjake, 2018-08-14:
```
[09:32:58][I][log_component:056]: Log initialized
[09:32:58][C][ota:182]: There have been 0 suspected unsuccessful boot attempts.
[09:32:58][I][application:048]: Running through setup()...
[09:32:58][C][wifi:031]: Setting up WiFi...
[09:32:58][I][wifi:227]: WiFi Connecting to 'MyWiFiSSID'...
[09:32:58][C][wifi:244]:     STA Hostname: 'lrb'
[09:33:08][I][wifi:278]: WiFi connected!
[09:33:08][C][wifi:280]:     SSID: MyWiFiSSID
[09:33:08][C][wifi:282]:     BSSID: [REDACTED]
[09:33:08][C][wifi:283]:     Channel: 1
[09:33:08][C][wifi:284]:     IP Address: 192.168.128.250
[09:33:08][C][wifi:285]:     Subnet: 255.255.255.0
[09:33:08][C][wifi:286]:     Gateway: 192.168.128.1
[09:33:08][C][wifi:287]:     DNS1: 205.171.3.26
[09:33:08][C][wifi:288]:     DNS2: 68.1.16.107
[09:33:08][C][ota:026]: Setting up OTA...
[09:33:08][C][ota:027]:     port: 3232
[09:33:08][C][ota:032]:     hostname: 'lrb'
[09:33:08][C][mqtt.client:024]: Setting up MQTT...
[09:33:08][C][mqtt.client:025]:     Server Address: MyMQTTBroker
[09:33:08][C][mqtt.client:026]:     Username: 'user'
[09:33:08][C][mqtt.client:027]:     Password: 'pass'
[09:33:08][C][mqtt.client:030]:     Client ID: 'lrb-b4e62db1eb31'
[09:33:08][C][mqtt.client:032]:     Discovery prefix: 'homeassistant'
[09:33:08][C][mqtt.client:033]:     Discovery retain: true
[09:33:08][I][mqtt.client:095]: Connecting to MQTT...
[09:33:10][I][mqtt.client:132]: MQTT Connected!
[09:33:10][C][esp32_ble_beacon:049]: Setting up ESP32 BLE beacon...
[09:33:10][D][debug:034]: esphomelib version 1.7.0ler
[09:33:10][D][debug:036]: Free Heap Size: 159908 bytes  failed: 259[D][debug:036]: Free Heap Size: 159908 bytes
[09:33:10]
[09:33:10][D][dGuru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[09:33:10]Core 0 register dump:
[09:33:10]PC      : 0x400d3568  PS      : 0x00060a30  A0      : 0x00000000  A1      : 0x3ffdb780
[09:33:10]A2      : 0xfde969a0  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
[09:33:10]A6      : 0x00000000  A7      : 0x00000000  A8      : 0xfde969a0  A9      : 0x3ffdb730
[09:33:10]A10     : 0x00000048  A11     : 0x3f401fc0  A12     : 0x3f402128  A13     : 0x3ffdb730
[09:33:10]A14     : 0x0000004e  A15     : 0x00000103  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
[09:33:10]EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
[09:33:10]
[09:33:10]Backtrace: 0x400d3568:0x3ffdb780 0x7ffffffd:0x3ffdb7e0
[09:33:10]
[09:33:10]Rebooting...
[09:33:10]ets Jun  8 2016 00:22:57
[09:33:10]
[09:33:10]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:33:10]configsip: 0, SPIWP:0xee
[09:33:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:33:10]mode:DIO, clock div:2
[09:33:10]load:0x3fff0018,len:4
[09:33:10]load:0x3fff001c,len:952
[09:33:10]load:0x40078000,len:6084
[09:33:10]load:0x40080000,len:7944
[09:33:10]entry 0x40080310
```
I believe I have debug turned on but I'm not really seeing anything different.

```
debug:

# Enable logging
logger:
  level: debug
```
Is what I have setup.

disrupted, 2018-09-27:
I am seeing the same issue also with `esp32_ble_tracker` component.

OttoWinter, 2018-09-27:
@disrupted Could you please post your debug output. It would help identify the issue for me :)

kluszczyn, 2018-09-27:
Hi, facing similar issue.

Config:
```
esphomeyaml:
  name: livingroom
  platform: ESP32
  board: nodemcu-32s
  arduino_version: latest
  
# Example configuration entry
esp32_ble_tracker:
  scan_interval: 300s

# + ble_rssi and xiaomi_miflora
```

Log:
```
[18:35:23][C][ota:027]: Setting up OTA...
[18:35:23][C][ota:028]:     port: 3232
[18:35:23][C][ota:033]:     hostname: 'livingroom'
[18:35:23][C][component:243]:     Update interval: 15000ms
[18:35:23][V][component:062]: set_interval(name='update', interval=15000, offset=3513)
[18:35:23][C][esp32_ble_tracker:112]: esp_bt_controller_init failed: 259y Level'...
[18:35:23][C][sensor.mqtt:031]:     Unit of Measurement: '%'ed: 259[C][sensor.mqtt:031]:     Unit of Measurement: '%'
[18:35:23]
[18:35:23][C][sensor.mqtt:032]:     Accuracy Decimals: 0
[18:35:23][C][sensor.mqtt:033]:     Icon: 'mdi:battery'

==========

[18:35:24][D][debug:033]: esphomelib version 1.8.0
[18:35:24][D][debug:035]: Free Heap Size: 153300 bytes
[18:35:24][D][debug:052]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[18:35:24][D][debug:082]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=0
[18:35:24][D][debug:084]: ESP-IDF Version: v3.2-dev-39-gaaf12390
[18:35:24][D][debug:087]: EFuse MAC: 0000F05803A4AE30
[18:35:24][D][debug:108]: Reset Reason: RTC Watch Dog Reset Digital Core And RTC Module
[18:35:24][D][debug:126]: Wakeup Reason: Unknown
[18:35:24][I][application:093]: Running through first loop()
[18:35:24][VV][component:136]: Running interval 'update':0 with interval=15000 last_execution=4294950057 (now=1535)
[18:35:24][V][sensor.sensor:026]: 'WiFi Signal Sensor': Received new value -56.000000
[18:35:24][D][sensor.mqtt:093]: 'WiFi Signal Sensor': Pushing out value -56.000000 with 0 decimals of accuracy
[18:35:24][I][application:109]: First loop finished successfully!
```

OttoWinter, 2018-09-27:
Related to https://github.com/nkolban/esp32-snippets/issues/592

Seems to be an issue on arduino framework's side. So maybe try with `arduino_version: dev` But I don't really understand from the linked issue what's actually causing this

kluszczyn, 2018-09-27:
With dev it's even stranger:
```
[19:44:33][C][ota:027]: Setting up OTA...
[19:44:33][C][ota:028]:     port: 3232
[19:44:33][C][ota:033]:     hostname: 'livingroom'
[19:44:33]E (1414) boot: Assert failed in igmp_lookup_group, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/igmp.c:269 (igmp_lookup_group: first group must be allsystems)
[19:44:33]
```

Path ``/Users/ficeto/Desktop/`` it's not from my setup

guardmedia, 2018-09-30:
I found I had to use a much older version of arduino-espressif32 to get this working. Has anyone else found a different solution?

```yaml
esphomeyaml:
  name: bttest
  platform: ESP32
  board: mhetesp32devkit
  arduino_version: espressif32@1.0.2
```

_Note: when reverting to 1.0.2, I had to delete AsyncTCP and allow esphomeyaml to download it again due to the IDF framework differences._

OttoWinter, 2018-10-04:
Thanks very much @guardmedia. I have created https://github.com/OttoWinter/esphomedocs/pull/34 to at least provide a workaround until the latest framework is fixed. The issue seems to be in the closed-source esp-idf framework, so esphomelib cannot do a workaround in software.

closed 2018-11-27 by OttoWinter