esphome/esphome #80: cant choose usb for upload

mefizto, 2018-07-09:
there is no thingy for choosing how to upload i cant upload, i cant upload firmware via USB

stefball, 2018-07-10:
If I remove and replug the cable in the USB port, the connection to serial port appears.

mefizto, 2018-07-10:
thank you stefball and zmrow

i already tried using different ports and different modules with no results, i can upload a firmware via arduino IDE.

I tried uploading the webupdate sketch and then upload the bin firmware but didn't work.

Next im going to try from Espeasy and then upload the esphomeyaml firmware. But it will be nice if i can just upload from USB, so i appreciate if you have more comments about it.

stefball, 2018-07-10:
The cable I used was directly connected to a port of a Raspberry 3 where I installed hassio.

mefizto, 2018-07-10:
i tried different ports and still can't choose the port

mefizto, 2018-07-11:
i finally use ESPeasy to upload the firmaware, it never showed me the option to upload to usb, none of the 4 ports of the raspberry worked, but I know that the ports work correctly

brandond, 2018-07-11:
Are you sure the Pi's USB ports are supplying sufficient power to the ESP32? If you're using it with an underpowered supply then it may power up enough to light the LEDs on the board, but not enough to actually operate either the ESP chip or the usb-serial converter that's built into the dev boards.

mefizto, 2018-07-13:
Sorry i was using the edge version of the add-on, I change to stable version and everything went fine.

closed 2018-07-16 by mefizto