esphome/esphome #81: docker aarch64 missing

stefball, 2018-07-09:
Trying to install extension from hassio the following error appears:
```
18-07-09 15:22:33 INFO (SyncWorker_19) [hassio.docker.interface] Pull image ottowinter/esphomeyaml-hassio-aarch64 tag 1.7.0.
18-07-09 15:22:35 ERROR (SyncWorker_19) [hassio.docker.interface] Can't install ottowinter/esphomeyaml-hassio-aarch64:1.7.0 -> 404 Client Error: Not Found ("manifest for ottowinter/esphomeyaml-hassio-aarch64:1.7.0 not found").
```
Verifying docker registry I found: 

No tags for this repository. :)

[image]

Gio76, 2018-07-13:
I'm getting the same issue:

```
INFO (SyncWorker_13) [hassio.docker.interface] Pull image ottowinter/esphomeyaml-hassio-aarch64 tag 1.7.0.
18-07-13 15:54:42 ERROR (SyncWorker_13) [hassio.docker.interface] Can't install ottowinter/esphomeyaml-hassio-aarch64:1.7.0 -> 404 Client Error: Not Found ("manifest for ottowinter/esphomeyaml-hassio-aarch64:1.7.0 not found").
18-07-13 15:54:59 INFO (SyncWorker_11) [hassio.docker.interface] Pull image ottowinter/esphomeyaml-hassio-aarch64 tag 1.7.0.
18-07-13 15:55:03 ERROR (SyncWorker_11) [hassio.docker.interface] Can't install ottowinter/esphomeyaml-hassio-aarch64:1.7.0 -> 404 Client Error: Not Found ("manifest for ottowinter/esphomeyaml-hassio-aarch64:1.7.0 not found").
18-07-13 15:55:34 INFO (SyncWorker_13) [hassio.docker.interface] Pull image ottowinter/esphomeyaml-hassio-aarch64 tag 1.7.0.
18-07-13 15:55:38 ERROR (SyncWorker_13) [hassio.docker.interface] Can't install ottowinter/esphomeyaml-hassio-aarch64:1.7.0 -> 404 Client Error: Not Found ("manifest for ottowinter/esphomeyaml-hassio-aarch64:1.7.0 not found").
18-07-13 15:55:42 INFO (SyncWorker_14) [hassio.docker.interface] Pull image ottowinter/esphomeyaml-hassio-aarch64 tag 1.7.0.
18-07-13 15:55:45 ERROR (SyncWorker_14) [hassio.docker.interface] Can't install ottowinter/esphomeyaml-hassio-aarch64:1.7.0 -> 404 Client Error: Not Found ("manifest for ottowinter/esphomeyaml-hassio-aarch64:1.7.0 not found").
```

This started after migrating to the new Raspberry Pi 3 64bit HassOS image.

brandond, 2018-07-13:
Looks like Docker doesn't know that aarch64 and arm64 are the same thing.

hiol83, 2018-07-16:
I've the same error

xtianpaiva, 2018-07-18:
Same error HassOS 1.7 64bits

frog32, 2018-07-18:
So far I was able to change to the edge version which is locally built. I did not yet check if everything works.

olskar, 2018-07-19:
@frog32 I also switched to edge version, but got #86 instead :)

ezar, 2018-07-19:
Please help us! we like this awesome utility :)

frog32, 2018-07-22:
Had time to give it a try with compiling something. Seems like some of the platform.io stuff is not fine with aarch64:

```
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Error: Could not find a version that satisfies the requirement '~1.40802.0' for your system 'linux_aarch64'
```

My opinion now is that it was a little early to go full blown 64 bit since there it probably not much benefit but lots of trouble involved.

OttoWinter, 2018-07-28:
Yes, I would love to be able to provide aarch64 images, but the problem is that platformio hasn't uploaded the precompiled build toolchain yet. So unfortunately RPi 3b/aarch64 support will have to wait until the platformio guys have figured that out. - I do not intend to try to build the toolchain myself, as I've had enough fun with debugging #19 ;-)

olskar, 2018-08-15:
Does this help @OttoWinter ?

> Toolchain has been deployed to our CDN

https://github.com/platformio/platform-teensy/issues/30

alpat59, 2018-09-08:
I have the same problem. HASS.io installed on Dietpi on Pine64 card. Everything works well. Many add-on installed. Only this one with installation problem. I have a really great interest on this project. Do you have some news about a possible fix in short term?
Thank you for your support

OttoWinter, 2018-09-20:
Yeah I'm also waiting for this, unfortunately I also have to wait for the platformio guys to upload a toolchain for the ESPs. I once tried to do the compiling myself but after two days of diagnosing weird library dependency problems and so on I gave up.

@olskar That issue is for teensy boards not ESPs so no luck there.

apeeters, 2019-01-20:
Platformio does not seem to have the intention to provide the espressif8266 toolchain for aarch64: [platformio/platformio-core#1263](https://sea-region.github.com/platformio/platformio-core/issues/1263) :(