esphome/esphome #82: ReadError: file could not be opened successfully

tarbax, 2018-07-09:
i get this error on 2 of my hassio instals when i want to upload the code:

```
LibraryManager: Installing esphomelib
git version 2.15.2
Cloning into '/config/esphomeyaml/vijver/.piolibdeps/_tmp_installing-Ur0iLZ-package'...
Note: checking out 'bd07644a80219761456b127cba7d92a514d73225'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

LibraryManager: Installing id=346 @ 0.8.2
Downloading...
Error: Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/usr/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/platformio/commands/run.py", line 108, in cli
    result = (envname, ep.process())
  File "/usr/lib/python2.7/site-packages/platformio/commands/run.py", line 207, in process
    result = self._run()
  File "/usr/lib/python2.7/site-packages/platformio/commands/run.py", line 302, in _run
    self.verbose)
  File "/usr/lib/python2.7/site-packages/platformio/commands/run.py", line 325, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "/usr/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/platformio/commands/lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "/usr/lib/python2.7/site-packages/platformio/managers/lib.py", line 399, in install
    force=force)
  File "/usr/lib/python2.7/site-packages/platformio/managers/lib.py", line 354, in install
    force=force)
  File "/usr/lib/python2.7/site-packages/platformio/managers/package.py", line 694, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/usr/lib/python2.7/site-packages/platformio/managers/lib.py", line 213, in _install_from_piorepo
    dl_data['url'], requirements)
  File "/usr/lib/python2.7/site-packages/platformio/managers/package.py", line 423, in _install_from_url
    self.unpack(dlpath, tmp_dir)
  File "/usr/lib/python2.7/site-packages/platformio/managers/package.py", line 216, in unpack
    with FileUnpacker(source_path) as fu:
  File "/usr/lib/python2.7/site-packages/platformio/unpacker.py", line 88, in __enter__
    self._unpacker = TARArchive(self.archpath)
  File "/usr/lib/python2.7/site-packages/platformio/unpacker.py", line 49, in __init__
    ArchiveBase.__init__(self, tarfile_open(archpath))
  File "/usr/lib/python2.7/tarfile.py", line 1680, in open
    raise ReadError("file could not be opened successfully")
ReadError: file could not be opened successfully
```

OttoWinter, 2018-07-28:
Whoa that's weird, I would suggest removing the generated platformio folder and retrying. Probably it's a network/platformio server issue. Unfortunately I can't do much about it since it's not an issue with esphomeyaml, but the platformio system.

OttoWinter, 2018-09-27:
I think this was an issue on platformio's servers, apparently they were sending out malformatted tar archives. Nothing esphomelib can fix and I think it was just momentary

closed 2018-09-27 by OttoWinter