esphome/esphome #83: Error compiling

mattcony, 2018-07-15:
Using Hass.io easyhomeyaml add-on. Any ideas for this error, trying to upload to an nodemcu-esp32s board with the basic .yaml file to my device.

Program failed with code 1

Error loading shared library libstdc++.so.6: No such file or directory (needed by /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++)
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_memory_used: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znwj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_newchunk: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: obstack_free: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdlPvj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _Znaj: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _obstack_begin: symbol not found
Error relocating /root/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: _ZdaPvj: symbol not found

OttoWinter, 2018-07-28:
Duplicate of #19 - Basically right now building firmwares for ESP32s using RPis is not possible.

OttoWinter, 2018-09-27:
Closing as duplicate...

closed 2018-09-27 by OttoWinter