esphome/esphome #88: ESP32 sends BME680 sensor values only once

cybrwushl93, 2018-07-18:
Hi,

first of all I really like this project! It makes using microcontroller for homeassistant a lot easier!
I'd like to use a ESP32 development board in combination with a BME680 sensor.
Unfortunately sends my ESP32 the sensor values only once over MQTT to my broker.

my configuration:
```yaml
esphomeyaml:
  name: officetemp
  platform: ESP32
  board: esp32dev

wifi:
  ssid: 'SSID'
  password: 'SECRET'

mqtt:
  broker: 'IP'
  username: ''
  password: ''
  topic_prefix: 'home/office'

# Enable logging
logger:

ota:
  password: 'OTA_SECRET'

i2c:
  sda: 21
  scl: 22
  scan: False

sensor:
  - platform: bme680
    temperature:
      name: "Office Temperature"
      oversampling: 16x
    pressure:
      name: "Office Pressure"
    humidity:
      name: "Office Humidity"
    gas_resistance:
      name: "Office Gas Sensor"
    address: 0x77
    update_interval: 30s
```
produces following output:
```log
[I][application:090]: Running through first loop()
[I][application:106]: First loop finished successfully!
[D][sensor.bme680:307]: Got temperature=25.6┬░C pressure=676.4hPa humidity=53.6% gas_resistance=12437703.0Ôäª
[D][sensor.mqtt:091]: 'Office Temperature': Pushing out value 25.618582 with 1 decimals of accuracy
[D][sensor.mqtt:091]: 'Office Pressure': Pushing out value 676.385803 with 1 decimals of accuracy
[D][sensor.mqtt:091]: 'Office Humidity': Pushing out value 53.615570 with 1 decimals of accuracy
[D][sensor.mqtt:091]: 'Office Gas Sensor': Pushing out value 12437703.000000 with 1 decimals of accuracy
[D][sensor.bme680:307]: Got temperature=25.7┬░C pressure=676.4hPa humidity=53.3% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.7┬░C pressure=676.5hPa humidity=52.8% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.7┬░C pressure=676.5hPa humidity=52.9% gas_resistance=12437703.0Ôäª
[I][ota:139]: Boot seems successful, resetting boot loop counter.
[D][sensor.bme680:307]: Got temperature=25.8┬░C pressure=676.5hPa humidity=52.8% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.8┬░C pressure=676.6hPa humidity=52.8% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.8┬░C pressure=676.6hPa humidity=52.3% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.9┬░C pressure=676.7hPa humidity=52.3% gas_resistance=12437703.0Ôäª
[D][sensor.bme680:307]: Got temperature=25.9┬░C pressure=676.7hPa humidity=52.9% gas_resistance=12437703.0Ôäª
```

brandond, 2018-07-18:
What you're seeing is covered in the docs here:
https://esphomelib.com/esphomeyaml/components/sensor/index.html#default-filter

cybrwshl, 2018-07-19:
Thanks for your reply, but this doesn't seem to work with the BME680...

```
[32mINFO [esphomeyaml.config] Reading configuration...[0m
Failed config
  sensor.bme680:
    - Invalid config for [sensor.bme680]: [filters] is an invalid option for [sensor.bme680]. Check: sensor.bme680->filters. (See .\officetemp.yaml, line 29).
    - platform: bme680
      address: 119
      filters: None
      gas_resistance: [source .\officetemp.yaml:38]
        name: Office Gas Sensor
      humidity: [source .\officetemp.yaml:36]
        name: Office Humidity
      pressure: [source .\officetemp.yaml:34]
        name: Office Pressure
      temperature: [source .\officetemp.yaml:31]
        name: Office Temperature
        oversampling: 16x
      update_interval: 30s
```

brandond, 2018-07-19:
Which release are you on?

cybrwshl, 2018-07-19:
nevermind I got it!

instead of using
```
sensor:
  - platform: bme680
    temperature:
      name: "Office Temperature"
      oversampling: 16x
      filters:
    pressure:
...
```
I tried:
```
...
    humidity:
      name: "Office Humidity"
    gas_resistance:
      name: "Office Gas Sensor"
    filters:
```

closed 2018-07-19 by cybrwshl